springfall2008/batpred #791: Two sunny day behaviours that are sub-optimal

PianSom, 2024-02-24:
**Describe the bug**
My Predbat plan displays two behaviours on sunny days that are sub-optimal. This has been observed several times this month. The first issue has also been reported by another user; the second seems to be only me.

**Expected behavior**
A more efficient plan should be followed

**Predbat version**
7.16.1

**Environment details**

- Inverter and battery setup
AIO plus pv panels
- Standard HAOS installer or Docker
Docker containers for HA, GivTCP, appdaemon (inc predbat as app)
- Anything else?

**Screenshots**
Today's behaviour:

[image]

Note the two sub-optimal behaviours:
 (1) the battery was charged to 100% overnight, even though there was a high solar generation forecast. It would have been cheaper/more efficient to charge to a lower level from the grid at 5.30am and then top up from surplus solar later in the morning
(2) having charged the battery to full/almost full from solar, Predbat chose to cycle around discharge/charge cycles. It would have been more efficient to freeze the battery and allow solar to carry the load and export any surplus.

I am on Intelligent Octopus Go (7.5p 11.30pm-5.30am, 30p otherwise) and Fixed export (15p).

Here is the plan, captured mid-morning:

[image]

**Log file**
No log file captured, but will do if necessary
Here is the latest backup of settings. Note that since then I have changed `input_number.predbat_metric_min_improvement_discharge` to ` 5.0`. I have also removed `export_limit` from apps.yaml (was 6000).

Please let me know if there is any personal info in this that I should not have posted - I don't think there is.

<details><summary>Settings backup</summary>
<p>

- default: false
  entity: update.predbat_version
  entity_picture: https://user-images.githubusercontent.com/48591903/249456079-e98a0720-d2cf-4b71-94ab-97fe09b3cee1.png
  friendly_name: Predbat Core Update
  installed_version: v7.16.1
  name: version
  release_url: https://github.com/springfall2008/batpred/releases/tag/v7.16.1
  restore: false
  title: Predbat
  type: update
  value: false
- default: false
  entity: switch.predbat_expert_mode
  friendly_name: Expert Mode
  name: expert_mode
  type: switch
  value: true
- default: false
  entity: switch.predbat_active
  friendly_name: Predbat Active
  name: active
  type: switch
  value: true
- default: 0.15
  entity: input_number.predbat_pv_metric10_weight
  friendly_name: Metric 10 Weight
  icon: mdi:multiplication
  max: 1.0
  min: 0
[... 2207 lines cut ...]
  type: input_number
  unit: days
  value: 0.0
- default: 24
  enable: expert_mode
  entity: input_number.predbat_forecast_plan_hours
  friendly_name: Plan forecast hours
  icon: mdi:clock-end
  max: 96
  min: 8
  name: forecast_plan_hours
  step: 1
  type: input_number
  unit: hours
  value: 24.0
- default: false
  enable: expert_mode
  entity: switch.predbat_plan_debug
  friendly_name: HTML Plan Debug
  name: plan_debug
  type: switch
  value: true

</p>
</details>

theblackcat9, 2024-02-25:
I'm on Octopus Flux tariff, Predbat is planning to charge my battery up to 96% tonight during the off-peak period at a cost including losses of 19.13p/kWh but this means with solar charging, the battery gets to 100% at 12:30/13:00 and starts to export to the grid during the day export rate of 14.86p/kWh including losses. I would have thought it was more profitable to charge the battery less overnight and let the solar charge the battery during the day reaching 100% at 4pm, the start of the peak export period and export the same amount during this peak period.
[image]
[image]
I've got the following parameters set:-
Battery loss charge 0.05
Battery loss discharge 0.05
Inverter loss 0.05
Best SOC keep 2.7
Best SOC min 0
Metric Battery cycle 0.5
Set Charge freeze off
[predbat_20240225.log](https://github.com/springfall2008/batpred/files/14396169/predbat_20240225.log)
[predbat_debug.txt](https://github.com/springfall2008/batpred/files/14396171/predbat_debug.txt)

springfall2008, 2024-02-25:
Can you show the battery curve with the 10% curve also?

I think the reasoning is that you want to maximise export at 24.97p including losses during that peak period. So the remaining solar (less your load) exported would be something like:

0.59 + 0.61 + 0.67 + 0.67 + 0.59 + 0.55 + 0.53 = 4.21
- 0.45 - 0.25 - 0.31 - 0.17 - 0.34 - 0.19  - 0.23 = - 1.94
= 2.27 exported

But if you account for the PV10% scenario I suspect most that margin disappears, hence the question on how that chart looks? You could also try pv10_weight of 0 to confirm my theory

PianSom, 2024-02-25:
There is a nice solar day forecast for Wednesday. Assuming that doesn't change (!) I will test then and revert.

theblackcat9, 2024-02-25:
Currently pv_metric10_weight is set to 0.15 and the Predbat plan for tonight is to charge to 100% which means with Solar, I start exporting to the grid at 12:00 tomorrow.

If I change pv_metric10_weight to 0, the plan changes to charge to 80% tonight which means with solar, the battery gets to 100% about 16:00 and starts exporting in the peak period, which is what I was after.

Sorry, I'm still quite new to Predbat, when you say "Can you show the battery curve with the 10% curve also?"  what exactly are you after?

PianSom, 2024-02-25:
Oops  @springfall2008 - were you talking  @theblackcat9 or me?

springfall2008, 2024-02-25:
> Currently pv_metric10_weight is set to 0.15 and the Predbat plan for tonight is to charge to 100% which means with Solar, I start exporting to the grid at 12:00 tomorrow.
>
> If I change [...]

So two things here:

1. The pv10 weighting is basically saying how probable you want Predbat to consider the more worst case scenario of lower PV generation (solcast 10%) and higher house load (load_scaling10). If you set this to say 0.1 then you are saying 10% or 1:10 probably. This means Predbat will optimise cost to 90% of the normal scenario and 10% of the worst case. If you set this to 0 your plan will be "cheaper" on paper but may turn out to be more expensive in reality if things don't go to plan 

2. I was meaning the apex chart that shows the battery levels, it shows the normal and the 10% scenario side by side

springfall2008, 2024-02-25:
> Oops  @springfall2008 - were you talking  @theblackcat9 or me?

Sorry I think i replied to black cat but the same is true for your 1.

For 2. Metric battery cycle will impact this, but change it gradually eg try in 0.5 increments

chrsphr, 2024-02-26:
I'm also experiencing behaviour that feels suboptimal. 

- 2kwp array
- 5kwh battery
- Agile tariff import/export
- I've fiddled with setting metric 10 to 0, 0.1, 0.2 and it didn't seem to make much difference.

My desired behaviour would be to maximise solar self consumption, especially given the export rates during the day are quite poor. 

Is there a parameter I am not quite setting correctly?

[image]

springfall2008, 2024-02-26:
> I'm also experiencing behaviour that feels suboptimal. 
> 
> 
> 
> - 2kwp array
> 
> - 5kwh battery
> 
> - Agile tariff import/export
> 
> - I've fiddled with setting metric 10 to 0, 0.1, 0.2 and it didn't seem to make much difference.
> 
> 
> 
> My desired behaviour would be to maximise solar self consumption, especially given the export rates during the day are quite poor, but on these sunnier days I've been exporting a kilowatt hour or so of solar when the battery has been charged to 100%
> 
> 
> 
> Is there a parameter I am not quite setting correctly?
> 
> 
> 
> [image]
> 
> 

To me this plan seems good, maximum self consumption and take the extra from the grid when it's really cheap to keep the battery for the peak period later on?!

theblackcat9, 2024-02-29:
Sorry for the delay in adding the charts, I've been waiting for a forecast with more solar as this shows a bigger difference.
With pv_metric10_weight set to 0.15, the overnight charge is to 100%, this means the battery gets back to 100% with solar at 12:00 and starts exporting to the grid:-
[image]

[image]

With pv_metric10_weight set to 0, the overnight charge is to 71%, this means the battery gets back to 100% with solar at 16:00 and then starts exporting to the grid:-
[image]
[image]

springfall2008, 2024-11-16:
Closing this old ticket, please open a new one if you have an up to date question