springfall2008/batpred #794: Option to reset cost to zero at midnight

pacemaker82, 2024-02-25:
**Describe the solution you'd like**
Switch on/off to have the predbat plan reset to zero at midnight so that its easier to understand the predicted cost day by day

**Describe alternatives you've considered**
I've built a custom card that extracts the final "best" prediction at 23:50 to help me understand the day to day costs, but its not ideal as not part of the plan. Would like to see the plan reset itself.

gcoan, 2024-02-25:
This could be added as an extra apps.yaml configuration if #782 is taken forward 

Eg
totals_reset_at_midnight: true

I'm suggesting that the extra configuration options be set in apps.yaml rather than HA switches as these will tend to be infrequently changed setup items, don't need to be changed dynamically by HA scripts and to reduce clutter of HA entities

gcoan, 2025-11-29:
> Did the reset feature ever get implemented? The issue is still open and #782 doesn't directly relate to this (unless you meant that the "reset" can be done in CSS overrides?)

No it hasn't been taken forward

Although it sounds a simple change, I think there's a lot of complexity in the detail, especially depending on how far you want to go with this.

782 was about making the predbat plan more configurable, and that's now closed as the predbat table card @pacemaker82 wrote gives loads of flexibility and configurability and is highlighted in the predbat documentation https://springfall2008.github.io/batpred/predbat-plan-card/#customising-and-reformatting-the-predbat-plan

That doesn't have the option to reset costs at midnight but I'm sure it could be fairly easily added so I have logged it https://github.com/pacemaker82/PredBat-Table-Card/issues/73

If all you want is to show the plan reset then this does the job.
If you want to change all the predbat entities and attributes so they show daily totals then (a) its a much bigger change and (b) it could well have undesirable consequences and mess up the predbat forward plan cost optimisation - which would need to be avoided

pacemaker82, 2025-11-29:
I will see what I can do on my custom table card to support this. Should be relatively easy... [famous last words]

gcoan, 2025-11-29:
I thought it might be.  Have raised an FR for you.

TBH I was surprised it wasn't an option already, seems an obvious thing to have, and would certainly be useful