springfall2008/batpred #795: Predbat crashed

SwiftRR, 2024-02-25:
**Describe the bug**
I noticed that my PredBat plan was not updating at 15:30 pm (not changed since 14:30).
I had been working on Charts so don't know whether I had broken anything.

I looked at my PredBat dashboard and noticed an exception:
Changed to ERROR: Exception raised Can't pickle <function wrapped_run_prediction_charge at 0x7f9b41f7e0>: it's not the same object as predbat.wrapped_run_prediction_charge

I restarted Home Assistant and eventually PredBat leapt back into action, much to my relief.
Before this Predbat was being shown as 'null'.
As Prdedbast reappeared, I noticed that the entities were all loading one by one from the Status information.
 
**Predbat version** 7.16.2 (set to automatically upload)

xxxx

**Environment details**

- Inverter and battery setup
- GE AC3.0 Coupled connected to two GE 8.2 batteries
- Standard HAOS installer or Docker
HA being run on a Raspberry Pi4 using a large SSD so no issues with ropy SD cards
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this). I have attached predbat log.

[predbat.log](https://github.com/springfall2008/batpred/files/14396873/predbat.log)

Thanks
Rob

springfall2008, 2024-03-03:
I suspect it relates to an update that didn't update the forked threads and restarting appdeamon would fix it. I should probably change the update code to wait for idle threads first or kill the pool...

springfall2008, 2024-03-03:
Fix pushed to main for the next release

closed 2024-03-07 by springfall2008