springfall2008/batpred #796: "Predbat active" shown in Control->Controls->Switches UI, not present in documentation

gundrygs, 2024-02-25:
A binary switch "Predbat active" is shown in the UI, but it's purpose and behaviour is not documented - I wonder if this should actually be part of the iBoost settings ?

Version v7.16.2 but has appeared in earler versions

**Environment details**

- GivEnergy 5.0 v2 inverver, GivEnergy 9.5Kw battery 
- Docker in VirtualBox on Mac mini

[image]

gcoan, 2024-02-25:
It was mentioned in the release notes.  Active is set to true when predbat is calculating, false otherwise.  You could use this to display a 'thinking' icon or similar on a dashboard, or to drive automations

I'll make sure its covered in the docs

gcoan, 2024-02-25:
It was introduced in patch #765

It is actually documented in the 'basic status' part of Output docs: https://springfall2008.github.io/batpred/output-data/#basic-status

gundrygs, 2024-02-25:
Thanks Geoffrey for that clear explanation - that explains why it seemed to be toggling on and off on it's own !
Useful, though the thought of automations ON TOP of Predbat rather makes my head spin !

I have closed the ticket, and sorry for my lack of care in perusing the excellent docs, made even better by your good self.

closed 2024-02-25 by gundrygs

gcoan, 2024-02-25:
No worries.

springfall2008, 2024-02-26:
Looks like you figured it out:)

It also serves a double purpose, if you toggle the switch it forces a recompute.

Although using the watch sensors maybe a better way to achieve this.