springfall2008/batpred #799: Nordpool issue

slopemad, 2024-02-27:
It looks like 'something' has changed at Nordpool and so the existing JSON file being downloaded is no longer being updated.  I am digging to see if there is a new location of this file...

slopemad, 2024-02-27:
OK, so looks like there is a new data format from a new URL:
https://dataportal-api.nordpoolgroup.com/api/DayAheadPrices?date=2024-02-28&market=N2EX_DayAhead&deliveryArea=UK&currency=GBP
The 'date' field is obviously going to be different every day (and if you search beyond the next day, it'll redirect to the latest dataset anyway).

slopemad, 2024-02-27:
Actually, it looks like the original data source has now updated, but it's probably worth noting that it may yet go away, but gives more time to look at the new datasource.

springfall2008, 2024-03-03:
Noted, although I don't know if these have the same data items as the other one had more EU prices?

springfall2008, 2024-11-16:
I think this was fixed, please re-open if not.

closed 2024-11-16 by springfall2008