springfall2008/batpred #800: Add support for HTML plan as array data in entity attribute

pacemaker82, 2024-02-28:
**Is your feature request related to a problem? Please describe.**
Related to https://github.com/springfall2008/batpred/issues/782 

**Describe the solution you'd like**
Would like a way to use the HTML plan information in a more systematic way. Ideally the HTML rows data could be stored as an array of objects (in time order) within a given entity.

E.g.

[0]
"time" : "Mon 09:00"
"import" : "7.5"
"export" : "15"
"limit" : 100

etc...

**Describe alternatives you've considered**
I have created my own HTML table view of the plan, but it scrapes the HTML and loads the data into arrays on the fly. This is a high risk approach as the HTML could change at any time from predbat plan, causing it to break. Getting the data from an entity array would be much less risky and performant.

**Additional context**
Please see my custom predbat table: https://github.com/pacemaker82/PredBat-Table-Card

gcoan, 2024-02-28:
I was thinking about this.

The current approach @pacemaker82 has done of re-parsing the HTML output is fragile and double handling the data, thus inefficient.

One option would be to add a new entity containing the data in a structured format which would then allow formatting/restructuring in a less fragile and more efficient way,  but at the cost of now duplicating the data from Predbat.

My proposal is a solution that brings together the best of both without the disadvantages.

The Predbat html plan be changed to be ONLY an output of data in a structured format, and the template code that is needed to render the HTML plan from the predbat output entity attribute be changed to render the plan as it stands at the moment, but from the structured data.

This way gives the ability for innovation on how the structured data is presented, means no double outputting of the data, and preserves the ability for users to display the plan using just "predbat" code rather than additional third party repositories - although these could be adopted if users wish.

This is clearly a breaking change, so could make this transition in a series of steps/releases:

1. Structured data output in addition to html plan output, announcement of when the html plan will be retired
2. Reformatter to format structured data to html plan output
3. Withdraw html plan output

This could cover some of the proposed changes in #782, but some may still be worth delivering as core predbat config options

chalfontchubby, 2026-10-19:
_Posted by Claude on behalf of @chalfontchubby._

Closing as already available: `predbat.plan_html` has a `raw` attribute holding the plan as unformatted JSON, documented in [output data](https://springfall2008.github.io/batpred/output-data/).

closed 2026-10-19 by chalfontchubby