springfall2008/batpred #801: Ability to specify multiple battery_scaling values

gcoan, 2024-02-29:
**Is your feature request related to a problem? Please describe.**
I have been running with just a single 5.2kW battery on one inverter for some time, so had set battery_scaling to 0.8 so that the 80% DoD SoC was correctly computed.

Had a new 9.5 commissioned (separate inverter) and wanted to continue with the 5.2 being scaled correctly, but Predbat gives a warning when I try to set multiple battery scaling values as the code must only be expecting a single value for all batteries across all inverters:

apps.yaml:

```yaml
  battery_scaling: 
    - 1.0
    - 0.8
```

logfile:

2024-02-29 08:30:00.103140 INFO pred_bat: WARN: Return bad float value [1.0, 0.8] from battery_scaling using default 1.0

**Describe the solution you'd like**
Ability to set battery scaling on a per inverter basis, allowing multiple entries as PV load, givtcp_rest, etc already do

**Describe alternatives you've considered**
Set a single hybrid battery scaling across all inverters, but this will give the wrong capacity result for both battery.

springfall2008, 2024-03-03:
Now on main, please test :)

gcoan, 2024-03-04:
Thanks Trefor, you been away on holiday, its been very quiet !

Loaded the version on main, set the battery scaling list in apps.yaml and from the logfile it appears to be working:

```
2024-03-04 01:15:12.583733 INFO pred_bat: Inverter 0 with soc_max 9.52 kWh nominal_capacity 9.52 kWh battery rate raw 2600 w charge rate 2.6 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 5.0 kW export limit 5.0 kW reserve 4.0 % current_reserve 6.0 %
2024-03-04 01:15:14.879243 INFO pred_bat: Inverter 0 SOC: 0.48 kW 5 % Current charge rate 2600.0 w Current discharge rate 2600.0 w Current power 0.0 w Current voltage 52.0
...
2024-03-04 01:11:07.200954 INFO pred_bat: Inverter 1 with soc_max 4.18 kWh nominal_capacity 5.22 kWh battery rate raw 2600 w charge rate 2.6 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 5.0 kW export limit 5.0 kW reserve 4.0 % current_reserve 6.0 %
2024-03-04 01:11:09.415656 INFO pred_bat: Inverter 1 SOC: 0.17 kW 4 % Current charge rate 2600.0 w Current discharge rate ``````

Inverter 0 isn't scaling the 9.5 battery whereas inverter 1 with the 5.2, it is scaling to 80%

I'll need to double check the kWh when the batteries are fully charged, but looks to be correct

BTW, I noticed the following warning in the log:

```
2024-03-04 01:11:05.882587 INFO pred_bat: Inverter 1 using Rest API http://homeassistant.local:6346
2024-03-04 01:11:07.190392 INFO pred_bat: WARN: Out of range index 1 within item inverter_battery_rate_min value 0
2024-03-04 01:11:07.192262 INFO pred_bat: Invertor time 2024-03-04 01:10:52+00:00 AppDaemon time 2024-03-04 
```

I don't have inverter_battery_rate_min set in my apps.yaml as its not needed for my config

springfall2008, 2024-03-04:
The warning is when you have a list of items but left it as default, I probably should fix that some time but it's unrelated.
Now in the latest release

Sadly I wasn't on holiday, was in the USA with work

closed 2024-03-04 by springfall2008

gcoan, 2024-03-04:
Thought you were away as it was so quiet, sorry it wasn't holiday.  Work gets in the way of having fun!

I just looked at predbat.soc_kw_h0 and it looked very wrong when compared to the underlying Soc values and kWh:
[image]

just as I was taking the screen shot, predbat finished an update run and soc_kw_h0 updated to a more reasonable value, so maybe it was just that it was 5 minutes out of date?

[image]

but still dropping nearly a kWh in 5 minutes even at full force discharging of both inverters seems unlikely. 2x 2.6kWh = roughly 1kWh discharged every 12 minutes or so.

I'll keep an eye on it