springfall2008/batpred #802: Passive-Mode (until other inverters are supported)

jimleach, 2024-02-29:
**Background**
I've built a HA setup for a friend who has Solar, battery, ASHP, car charging Zappi, myenergy, Octopus Agile (you name it!) - the inverter is a _Sunsynk_, so not yet supported by Predbat but we're getting all the data from the add-on: https://kellerza.github.io/sunsynk/

**Describe the solution you'd like**
Would it be possible to have a Passive-mode (similar to monitor) whereby Predbat can take all the data inputs, but with no output (ie not configured for or expecting to send to GivTCP or other inverters) _or_, build a set of binary toggles which Predbat changes according to if the slot is a discharge/charge/hold slot.   The idea being we can then build automations around the generic output rather than you having to build specific integrations for each inverter?

gcoan, 2024-02-29:
If you set predbat to the mode you want (eg charge and discharge), and then read only mode, it'll do what you want.

Trouble is that at the moment it expects to be connected to at least one inverter type it knows about for this to work.

You may be able to get it to work with GivEnergy inverter type, non-rest mode, and dummying key GivTCP entities that predbat needs such as max charge rate, battery soc, battery size, etc.

If these key fields were identified in apps.yaml and the documentation it'd be easier.  Not sure we want a passive mode that every user will see and could accidentally chose from, but a passive inverter type instead would work.

jimleach, 2024-02-29:
It's this - I couldn't see a way of enabling / configuring it without an inverter... so really it needs a "dummy" inverter which can be fed the inputs from an unsupported inverter

gcoan, 2024-02-29:
Agreed, a dummy inverter type that you can set in apps.yaml and then set the minimum required set of sensors that Predbat needs

springfall2008, 2024-03-03:
I think one of the simpler options like HU or SF should work with dummy apps.yaml and using Read Only or Monitor mode. Try put together a configuration and see where you get to, I can always add Sunsyk options then?

springfall2008, 2024-03-15:
I've now made changes so you can add your own inverter configuration in apps.yaml directly, please re-read the other inverter docs section

jimleach, 2024-03-16:
Hi - thanks for doing this.. Did you commit the change though and would that commit be to the add-on version?  I have followed to the best of my ability the "Other inverter guide" - my app.yaml looks like this:

**Sensor input**
```
  load_today:
    - sensor.ss_day_load_energy
  import_today:
    - sensor.ss_day_grid_import
  export_today:
    - sensor.ss_day_grid_export
  pv_today:
    - sensor.ss_day_pv_energy
```

**Inverter Config**

```
  inverter_type: "MINE"
  inverter:
    name : "SYNK-5K-SG04LP1"
    has_rest_api: False
    has_mqtt_api: False
    has_service_api: false
    output_charge_control: "power"
    has_charge_enable_time: False
    has_discharge_enable_time: False
    has_target_soc: False
    has_reserve_soc: true
    charge_time_format: "S"
    charge_time_entity_is_option: False
    soc_units: "%"
    num_load_entities: 1
    has_ge_inverter_mode": False
    time_button_press: False
    clock_time_format: "%Y-%m-%d %H:%M:%S"
    write_and_poll_sleep: 2
    has_time_window: False
    support_charge_freeze: False
    support_discharge_freeze": False
```

**other inverter/sensors**

```
  battery_power:
    - sensor.ss_battery_power
  pv_power:
    - sensor.ss_pv1_power
    - sensor.ss_pv2_power
  load_power:
    - sensor.ss_essential_power
  soc_percent:
     - sensor.ss_battery_soc
  # soc_max ** Have assumed this is in percent **
  soc_max:
     - 10
  # reserve ** Have assumed this is in percent **
  reserve:
    - 20
  # charge_limit ** Have assumed this is in percent **
  charge_limit:
    - 100
  scheduled_charge_enable:
   - off
  scheduled_discharge_enable:
  - off
```

**It looks like the add-on version runs an older code, even though I only added/downloaded it this morning**

```
2024-03-16 10:28:44.467373 INFO pred_bat: Sanity: /config/apps/apps.yaml is a valid pred_bat configuration
2024-03-16 10:28:44.470718 INFO pred_bat: Sanity: Confirmed correct version v7.15.15 is in predbat.py
2024-03-16 10:28:44.479789 INFO pred_bat: Sanity check has passed
```

**The log when starting the appdaemon-predbat add-on is as follows:**

```
2024-03-16 12:34:43.232703 WARNING pred_bat: ------------------------------------------------------------
2024-03-16 12:34:43.233522 WARNING pred_bat: Unexpected error running initialize() for pred_bat
2024-03-16 12:34:43.234588 WARNING pred_bat: ------------------------------------------------------------
2024-03-16 12:34:43.238569 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
    await utils.run_in_executor(self, init)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 11850, in initialize
    self.log(f"Inverter Type: {self.inverter_type} ({INVERTER_TYPES[self.inverter_type]})")
                                                     ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
KeyError: 'MINE'
```

gcoan, 2024-03-16:
It looks like you are on an old version of predbat.

The combined predbat app daemon version is updated less frequently, the normal approach is to let the predbat auto update install the new version automatically but looks like predbat isnt getting that far and is failing because it doesn't recognise the inverter type

Either needs a newer version of the combined install or if you can change apps.yaml to an inverter type that is recognised it should then run through to the auto update to install the predbat version you need

jimleach, 2024-03-16:
I thought so to - so I have replaced predbat.py with the latest version, but now I'm getting an error which makes me thing I've put the wrong unit in the config somewhere:

(I've clipped this error as it's long)
```
2024-03-16 12:55:27.089582 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 12641, in update_time_loop
    raise e
  File "/config/apps/predbat.py", line 12636, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 11656, in update_pred
    status, status_extra = self.execute_plan()
                           ^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 10596, in execute_plan
    inverter.adjust_discharge_rate(inverter.battery_rate_max_discharge * MINUTE_WATT)
  File "/config/apps/predbat.py", line 2923, in adjust_discharge_rate
    entity = self.base.get_entity(self.base.get_arg("discharge_rate", indirect=False, index=self.id))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```
`TypeError: argument of type 'NoneType' is not iterable`

This is the config for battery_rate_max:
```
  # battery_rate_max ** assume this is kWh **
  battery_rate_max:
    - 5000
```

gcoan, 2024-03-16:
It's discharge rate it can't find in apps.yaml.
Should be a number in watts eg 2600 for 2.6kW

jimleach, 2024-03-16:
Right, thanks for all your help/pointers - seem to have got it all working now- going to let it run in Monitor mode to see how it behaves while I work out how to control the charge schedule on the Sunsynke

gcoan, 2024-03-16:
Great, no problem.  When its all working if you can share any advice/config/instructions I can add it to the documentaton

Cheers

springfall2008, 2024-11-16:
Closing this old ticket, please open a new one if you have an up to date question

closed 2024-11-16 by springfall2008