springfall2008/batpred #803: Charging early - 30 mins before powerup

onetoomany, 2024-02-29:
We're currently in a powerup (12:00 -> 15:30). My battery discharged down to 20% for 11.30, then BatPred decided that exactly 11.30 was a good time to start to charge the battery, rather than sit idle for 30mins while waiting for the powerup at 12:00 when rates would be 0.

version v7.16.2
integration via sofar mqtt
[log.zip](https://github.com/springfall2008/batpred/files/14447965/log.zip)

   -  date: '2024-02-29'
      start: '12:00:00'
      end: '15:30:00'
      rate: 0

gcoan, 2024-02-29:
What’s your min soc values (best_Soc_keep and Best_Soc_min)?Wonder if this was triggering the extra charge?I didn’t have a similar issue, Predbat discharged the batteries, one fully and one to almost empty just before the session and they’re now recharging

onetoomany, 2024-02-29:
These are set at:  

soc_kw:
    - sensor.sofar_battery_energy
  soc_max:
    - input_number.sofar_battery_max_charge
  charge_limit:
   - 100
  reserve:
    - 20

gcoan, 2024-02-29:
Thanks. I was asking about inpit_number.predbat_best_soc_keep and_min which are set in Home Assistant.

These control the minimum level the battery can discharge down to

onetoomany, 2024-02-29:
predbat_best_soc_keep = 2.0
predbat.best_soc_min_kwh = 0.96 (aka 4.8kwh battery * 20%)

gcoan, 2024-02-29:
Ah, I think that's the cause of your problem.

Predbat has two different important battery limits

https://springfall2008.github.io/batpred/customisation/#battery-margins-and-metrics-options

Input_number.predbat_best_soc_keep is a soft limit that Predbat tries to keep your battery above, but input_number.predbat_best_soc_min is a hard limit that Predbat will always ensure your battery remains above this limit.

If you had _min set to 20% of your battery level, then as soon as your battery drops below this level, Predbat will charge the battery, regardless of the rate.

_min is basically your backup reserve amount you don't normally want to go below, e.g. if you have emergency power supply capability from your battery and inverter, keep some in reserve in case of power cut. You should set this as low as possible, e.g. I have mine set to 0 and best_soc_keep to 20% of my battery level.

onetoomany, 2024-02-29:
not sure. The charging started at 11.30 when the battery charge was at 25%; it then charged at full rate until 12:00 at 53%. At this point, the powerup started so the battery started to discharge (!).

springfall2008, 2024-03-03:
I think there is a bug in the inverter control here, Predbat reports Idle but its charging

springfall2008, 2024-03-03:
I've pushed a fix to 'main' can you try the version there please (v7.16.3) if it works I'll release it?

onetoomany, 2024-03-04:
great - I'll give this fix a go. Many thanks.
I'll keep an eye on it over the next few days and see how it behaves.

onetoomany, 2024-03-12:
this looks like its working fine with the fix - currently monitoring it during today's powerup without any issue.
@springfall2008  - many thanks!

closed 2024-03-12 by onetoomany