springfall2008/batpred #806: Charging before cheap rate for no reason

WillCalderwood, 2024-02-29:
**Describe the bug**
There's a saving session today, and PredBat is holding charge for that, but it's then charging after the saving session, for which I can see no reason.

[image]

**Expected behaviour**
Don't charge at 4x the price when there's no benefit.

**Predbat version**
7.16.2

**Environment details**

- Inverter and battery setup
GivEnergy inverter and battery

- Standard HAOS installer or Docker
VirtualBox install

[predbat-log (2).txt](https://github.com/springfall2008/batpred/files/14450600/predbat-log.2.txt)

gcoan, 2024-02-29:
Agreed, it's charging enough after the saving session to ensure your battery doesn't run out until after the start of the overnight cheap period, but after losses are included, this isn't optimal.

What are your loss percentages set to and input_number.predbat_best_soc_min and predbat_best_soc_keep?

This isn't happening on my predbat, it just lets the battery discharge after the saving session, so suggests its something in your config that's causing it to think its worthwhile to charge

WillCalderwood, 2024-02-29:
My settings are mostly defaults.
Charge Loss 3%
Discharge Loss 3%
Inverter Loss 4%

I'm not sure where to find predbat_best_soc_min and predbat_best_soc_keep.

gcoan, 2024-02-29:
If you look in settings / integrations / entities you can search for these two.

They are prefixed input_number.predbat_...

WillCalderwood, 2024-02-29:
predbat_best_soc_min = 0
 predbat_best_soc_keep = 0.5

gcoan, 2024-02-29:
Thanks Will,  your losses are a bit on the low side,  but even so, Predbat still should have thought it wasn’t profitable to charge when it did.Your best Soc values are not what’s causing the planned charging so sorry I can’t see anything obviously wrong.Can you attach the logfile that covers the period when the plan was created, or even anything up to before it started the charging after the saving session.  Trefor might be able to see what caused it

WillCalderwood, 2024-02-29:
I included the log file in the first post. Thanks.

springfall2008, 2024-03-03:
The reason for this charge is an artifact of the change I made to the way soc keep works, now if you have a discharge then it must obey the keep value, but this means the algorithm has planned a charge to follow the rules. What you will find is the charge will never happen as once the discharge completes the keep won't need to be followed for the next 4 hours and as that time never comes it wont even happen (until cheap rate kicks in)

springfall2008, 2024-11-16:
Closing this old ticket, please open a new one if you have an up to date question

closed 2024-11-16 by springfall2008