springfall2008/batpred #807: Actual cost daily/weekly graph dispaly - request

timparker71, 2024-02-29:
It would be great if one of the cards/graphs could show the actual cost at the end of each day and maybe display the last weeks worth of daily actuals.  I view mine by checking the history at 23:59. Hopefully most people keep at least seven days worth of data to call on, I keep a months worth, but something showing the last seven days 'actual' would be great!
Thank you.

gcoan, 2024-02-29:
If you set your import and export rates up and your import and export sensors on the Energy dashboard in Home Assistant, it’ll do all this for you and you can view consumption, battery charge/discharge, import and export costs by day, week, month, etc.If you are using the Octopus Energy integration (bottlecapdave) then there’s instructions for how to use the rates supplied by Octopus in the github documentation for the integration

Nilogax, 2024-02-29:
I've done this using a template "Daily Cost" sensor from {{states('predbat.cost_today')}}, then use a statistics-graph card to show values from the last N days

7Van737, 2024-02-29:
If you have Apex custom cards installed then paste the following code into a card and it will show you a 14 day graph of daily electric costs

type: custom:apexcharts-card
graph_span: 14d
span:
  end: day
header:
  show: true
  title: Daily Cost
  show_states: true
  colorize_states: true
series:
  - entity: predbat.cost_today_import
    type: column
    name: /day
    group_by:
      func: last
      duration: 1d

timparker71, 2024-02-29:
Thank you for this @7Van737. I've created the card, but it doesn't show me the actual cost at the end of the day, ie at 23:59. Sometimes for me this is a negative number, sometimes it's pence. Thanks

timparker71, 2024-02-29:
This is the figure I'd like at the end of each day:
[image]

timparker71, 2024-02-29:
@7Van737 I've fixed it! Simple! Thanks again
[image]

7Van737, 2024-02-29:
I don't currently have an export MPAN so don't export - hence I don't have any negative costs.   I guess you've changed the entity to use  predbat.cost_today   ??

gcoan, 2024-02-29:
I can add this to the sample graphs so other users can benefit from it, thanks @Nilogax @7Van737 

It is worth getting the Energy card working as well though as its much better at looking at longer term consumption trends/graphs - although annoyingly it only shows total cost for the time period eg for last week, not broken down cost per day