springfall2008/batpred #808: SoC fluctuate around 100%, expected plan behaviour?

Jingbin-GitHub, 2024-03-01:
**Describe the bug**
When the import rate is cheap, battery is set to charging to 100% for a big slots. This makes sense, so we use grid power rather than the battery.

But instead of using the grid power, the battery discharge to support the load to 96% and than charge back up to 100% and discharge again. I am expecting the battery to stay at 100% and all load is met by grid. Even the hold charge at 100% having the same behaviour on the SoC, i.e. fluctuating. 

**Expected behavior**
In such situation, I expect that that battery is set to 'idle' (does not charge or discharge).

**Predbat version**

7.16.2

**Environment details**

- Inverter and battery setup - AIO
- Standard HAOS installer or Docker, standard HAOS
- Anything else?

**Screenshots**
[image]
[image]
![IMG_3089](https://github.com/springfall2008/batpred/assets/154021101/76cc78
[image]
3f-b9c2-4c87-9783-642ab19aa02e)

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

springfall2008, 2024-03-03:
I suspect its due to the max reserve being below 100% (AIO seems to not allow a reserve of 100%) and hence the battery still discharging, although I thought I had fixed that issue. I would need a logfile to debug it, can you capture one next time?

Jingbin-GitHub, 2024-03-03:
added SoC graph for this morning and the log file (3rd Mar). Thanks!
[image]
[appdaemon.txt](https://github.com/springfall2008/batpred/files/14474372/appdaemon.txt)

springfall2008, 2024-03-04:
Something is setting the reserve to 10% whenever Predbat sets it to 100%

```
2024-03-03 01:20:11.692403 INFO pred_bat: Inverter 0 Current Reserve is 10.0 % and new target is 100 %
2024-03-03 01:20:21.696782 INFO pred_bat: Inverter 0 Wrote 100 to reserve, successfully now 100
```

Can you try to set in apps.yaml:

```yaml
inverter_reserve_max : 98
```

I think this will resolve the issue

springfall2008, 2024-03-04:
BTW: If you had REST enabled you might have seen an error here...

Jingbin-GitHub, 2024-03-05:
Having the inverter_reserve_max set to 98 solves the problem! It now behaves exactly how I would expect it to be. Thanks for your help! Really appreciated.

I don't have REST API enabled.

springfall2008, 2024-03-05:
Great :) I'd recommend turning on REST unless you have a specific reason not to. Glad its solved

closed 2024-03-05 by springfall2008

springfall2008, 2024-03-05:
https://github.com/springfall2008/batpred/pull/822