springfall2008/batpred #810: Strange behaviour prior to Octopus saving session

Sycamore967, 2024-03-03:
**Describe the bug**
Octopus ran a saving session on 02/03/24 @18:00 Predbat joined the session ok but set a plan prior to it saying No Charge.  When I demanded over a 2Kwh load (microwave) Predbat didn;t allow the battery to discharge to meet that load and I noticed the import from the grid.  To quickly get around it, I used force discharge for 15 mins.  I have rates_export_ override set between 17:00-19:00 is there some sort of planning conflict?

**Expected behavior**
I expected the battery to meet my load demand

**Predbat version**
v7.16.2

**Environment details**

Gen 3 inverter 2x 9.5Kwh batteries

**Screenshots**
Sorry didn't take any,

**Log file**
I've attached apps & appsdaemon.log1 that covers the time period.
[app logs.zip](https://github.com/springfall2008/batpred/files/14472583/app.logs.zip)

springfall2008, 2024-03-03:
Seems reasonable to me, Predbat held the battery just before the saving session so you had enough to complete the export and stick within your metric keep which is set at 1.9kWh

Sycamore967, 2024-03-04:
Yeah that's fine, but the export window was 30 minutes and at the start of it my SOC was 12.95kWh - it was only ever going to export <2kWh and there was more scheduled exports later that night.  If I had 3 or 4 kWh I could undertand it protecting the SOC but this seems a bit cautious?

Sycamore967, 2024-03-05:
Hi, yes that makes sense, I'll make the changes and see how I get on.  Thanks.

Sycamore967, 2024-03-29:
Ok just a quick update and what I've changed.

Setting combine_charge, to false didn't work well.  It really made the plan go all over the place, so I've reverted this change.

Setting set_discharge_during_charge set to True (default).  This is working well.  The plan makes sense and I've seen no issues with the single  Saving Sessions since.

closed 2024-03-29 by Sycamore967