springfall2008/batpred #819: Add currency symbol option

springfall2008, 2024-03-04:
**Is your feature request related to a problem? Please describe.**
Predbat shows prices in £s

**Describe the solution you'd like**
Add option to change to $'s, Euro's etc.

**Describe alternatives you've considered**
Just ignore it

gcoan, 2024-03-04:
should predbat not just pick the currency symbol from Home Assistant rather than having to have yet another manual config item in apps.yaml?

springfall2008, 2024-03-06:
Now implemented on main ready for testing, please note I have no idea how to link this to HA directly as its via AppDeamon, but at least you can configure this in apps.yaml now

gcoan, 2024-03-06:
I'd forgotten that HA and appdaemon are running as different worlds without a lot of knowledge of each other.

Had a Quick Look and I can't find anything on google that suggests how to do this and its not something that is set in appdaemon either https://appdaemon.readthedocs.io/en/latest/CONFIGURE.html

so a local config in apps.yaml at least makes the capability available to Predbat for our overseas users

Cheers

closed 2024-03-17 by springfall2008