springfall2008/batpred #826: Lost wi-fi, Predbat didn't fail in a detectable way

gcoan, 2024-03-06:
**Describe the bug**
The wi-fi node in my garage lost connection this afternoon so the inverters went offline.
Predbat started reporting errors with time being incorrect and REST failing & would be disabled, but because predbat.status changed every 5 minutes the script I have to detect a Predbat failure occuring for a period of time couldn't be triggered.

history of predbat.status:
[image]
[image]
[image]
and carries on like this for the next 2 hours
[image]

**Expected behavior**
1. Failure occurred reading from REST, Predbat says it will stop using REST, but 5 minutes later it tries again. If REST fails then Predbat should stop trying to use it (or at least stop for a decent time period, e.g. 30 minutes).
2. Once an error occurs, leave the status showing the error, don't change it.  This means an external automation to detect "predbat.status contains 'ERROR' for 10 minutes" can be triggered

I suspect the REST issue is what caused the second issue.  Might be that the errors were coming from the balance inverter code?

**Predbat version**
7.16.3

**Environment details**
HAOS, 2x 5kW hybrid inverters

**Log file**
logfile extract:
```
15:25:01.729020 INFO pred_bat: WARN: Historical day 8 has 40 minutes of gap in the data, filled from 48.62 kWh to make new average 50.01 kWh (percent 97%)
2024-03-06 15:25:01.736723 INFO pred_bat: Historical days now [2, 4, 5, 6, 7, 8] weight [1, 1, 1, 1, 1, 1]
2024-03-06 15:25:01.763217 INFO pred_bat: Today's load divergence 76.36 % in-day adjustment 76.36 % damping 1.0x, Predicted so far 24.3 kWh with 0.0 kWh car excluded and 13.72 kWh import ignored and 0 forecast extra, Actual so far 16.13 kWh with 0.0 kWh car excluded and 5.03 kWh import ignored
2024-03-06 15:25:01.849895 INFO pred_bat: Inverter 0 using Rest API http://homeassistant.local:6345
2024-03-06 15:25:19.248758 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-03-06 15:25:19.269353 INFO pred_bat: WARN: Inverter control auto restart trigger: REST read failure
2024-03-06 15:25:19.271909 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-03-06 15:25:19.274717 INFO pred_bat: Inverter 0 with soc_max 10.0 kWh nominal_capacity 10.0 kWh battery rate raw 2600.0 w charge rate 2.6 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 5.0 kW export limit 5.0 kW reserve 4.0 % current_reserve 4.0 %
2024-03-06 15:25:30.390865 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-03-06 15:25:30.412633 INFO pred_bat: Inverter 0 SOC: 9.52 kW 95 % Current charge rate 2600.0 w Current discharge rate 0.0 w Current power 0.0 w Current voltage 52.0
2024-03-06 15:25:30.414881 INFO pred_bat: WARN: Inverter 0 unable to read charge window time as neither REST, charge_start_time or charge_start_hour are set
2024-03-06 15:25:30.415634 INFO pred_bat: ERROR: Exception raised cannot access local variable 'charge_start_time' where it is not associated with a value
2024-03-06 15:25:30.461134 INFO pred_bat: BALANCE: Enabled balance charge False discharge False crosscharge True threshold charge 1.0 discharge 1.0
2024-03-06 15:25:52.306228 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-03-06 15:25:52.324098 INFO pred_bat: WARN: Inverter control auto restart trigger: REST read failure
2024-03-06 15:25:52.326758 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-03-06 15:26:03.355603 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-03-06 15:26:03.384937 INFO pred_bat: WARN: Inverter 0 unable to read charge window time as neither REST, charge_start_time or charge_start_hour are set
2024-03-06 15:26:03.385538 INFO pred_bat: ERROR: Exception raised cannot access local variable 'charge_start_time' where it is not associated with a value
2024-03-06 15:26:15.008530 INFO pred_bat: BALANCE: Enabled balance charge False discharge False crosscharge True threshold charge 1.0 discharge 1.0
2024-03-06 15:26:26.105963 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-03-06 15:26:26.128181 INFO pred_bat: WARN: Inverter control auto restart trigger: REST read failure
2024-03-06 15:26:26.130998 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-03-06 15:26:37.209918 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-03-06 15:26:37.232383 INFO pred_bat: WARN: Inverter 0 unable to read charge window time as neither REST, charge_start_time or charge_start_hour are set
2024-03-06 15:26:37.232990 INFO pred_bat: ERROR: Exception raised cannot access local variable 'charge_start_time' where it is not associated with a value
2024-03-06 15:27:15.007148 INFO pred_bat: BALANCE: Enabled balance charge False discharge False crosscharge True threshold charge 1.0 discharge 1.0
2024-03-06 15:27:30.318548 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-03-06 15:27:30.342272 INFO pred_bat: WARN: Inverter control auto restart trigger: REST read failure
2024-03-06 15:27:30.345262 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-03-06 15:27:41.439419 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-03-06 15:27:41.460698 INFO pred_bat: WARN: Inverter 0 unable to read charge window time as neither REST, charge_start_time or charge_start_hour are set
2024-03-06 15:27:41.461634 INFO pred_bat: ERROR: Exception raised cannot access local variable 'charge_start_time' where it is not associated with a value
2024-03-06 15:28:15.007752 INFO pred_bat: BALANCE: Enabled balance charge False discharge False crosscharge True threshold charge 1.0 discharge 1.0
2024-03-06 15:28:34.662200 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-03-06 15:28:34.682252 INFO pred_bat: WARN: Inverter control auto restart trigger: REST read failure
2024-03-06 15:28:34.685500 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-03-06 15:28:45.815891 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-03-06 15:28:45.837641 INFO pred_bat: WARN: Inverter 0 unable to read charge window time as neither REST, charge_start_time or charge_start_hour are set
2024-03-06 15:28:45.838245 INFO pred_bat: ERROR: Exception raised cannot access local variable 'charge_start_time' where it is not associated with a value
2024-03-06 15:29:15.008233 INFO pred_bat: BALANCE: Enabled balance charge False discharge False crosscharge True threshold charge 1.0 discharge 1.0
2024-03-06 15:29:26.078280 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-03-06 15:29:26.096959 INFO pred_bat: WARN: Inverter control auto restart trigger: REST read failure
2024-03-06 15:29:26.099483 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
[... 385 lines cut ...]
2024-03-06 15:41:37.407173 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-03-06 15:41:37.422026 INFO pred_bat: WARN: Inverter 0 unable to read charge window time as neither REST, charge_start_time or charge_start_hour are set
2024-03-06 15:41:37.422631 INFO pred_bat: ERROR: Exception raised cannot access local variable 'charge_start_time' where it is not associated with a value
2024-03-06 15:42:15.058780 INFO pred_bat: BALANCE: Enabled balance charge False discharge False crosscharge True threshold charge 1.0 discharge 1.0
2024-03-06 15:42:30.133758 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-03-06 15:42:30.160000 INFO pred_bat: WARN: Inverter control auto restart trigger: REST read failure
2024-03-06 15:42:30.163238 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-03-06 15:42:41.245431 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
[... cut ...]
2024-03-06 15:44:26.191781 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-03-06 15:44:48.186829 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2024-03-06 15:44:48.203912 INFO pred_bat: WARN: Inverter 0 unable to read charge window time as neither REST, charge_start_time or charge_start_hour are set
2024-03-06 15:44:48.204740 INFO pred_bat: ERROR: Exception raised cannot access local variable 'charge_start_time' where it is not associated with a value
```

springfall2008, 2024-03-07:
I've pushed a fix to main to:

1. Catch the crash with charge_start_time
2. Created a new 'error' attribute in predbat.status which is True when the system is in error. I think if you base your script on that rather than the text in status then you might have more luck?

gcoan, 2024-03-07:
thanks, I'll give it a test and see how it behaves.  Is the error attribute set only on this error or is it more general set when there's a predbat failure, and I assume the attribute is cleared if Predbat gets to the end of a normal run?

springfall2008, 2024-03-08:
It's just a binary true/false for any type of error so it should be false for normal runs

gcoan, 2024-03-16:
thanks, I have updated my automation to use this error attribute and added details of using it to the documentation

closed 2024-03-16 by gcoan