springfall2008/batpred #829: Cross charging during charging event

gcoan, 2024-03-07:
**Describe the bug**
#138 was raised last year about a cross charging issue I was seeing with my two inverters, in particular during charging where if one soc was higher than the target the higher Soc inverter+battery would discharge into the lower.

This was addressed by turning off discharge during charge event.  The issue concluded with me saying I needed to "use the 'balance inverters for cross charging=on' and 'set discharge during charge=off' switches to prevent cross charging"

I have these set:
[image]
[image]

But towards the end of the power up event today I saw cross charging occurring:
[image]

Battery power for G in blue remains negative almost all the way throughout, but H in purple goes from negative to positive and then back and forth as H discharges to G from 14:42 to 15:00.

Discharge rate for H remained on 2600 (and per 138 should have been set to zero) for almost all of the period.  The two changes to zero at the end were me manually invervening to set the discharge rate and then put predbat into read only.
[image]

As we got towards the end of the charging period Predbat calculated that the batteries wouldn't be full in time (I'll raise a separate defect on this) and started reducing the target SoC.  This corresponded to the cross charging starting.
[image]

**Expected behavior**
During a charging period no cross charging occurs. Either the 'set discharge during charge' switch be reinstated to stop one battery discharging to the other when both are supposed to be charging, or the target socket doesn't get revised downwards.

**Predbat version**
7.16.3

**Environment details**
HAOS
2x GivEnergy 5kW Hybrid inverters, one with a 5.2 and one with a 9.5 battery

springfall2008, 2025-01-01:
As this ticket is old I'm going to close it, please re-open if the issue is there on latest.

gcoan, 2025-01-01:
Appreciate you clearing the clutter out Trefor

As we had a powerup event today from 9-12 it's easy for me to see what the behaviour of the two inverters was.

Battery power, predbat status and SoC:

[image]

There are a couple of anomalies, at about 9:25 and 10:00 first one then the other inverter stopped charging despite both being a long way from 100% full.

From about 10:10 on H and 10:50 on G I see the 10-step discharge slowdown as each battery nears to 100% - this is normal for BMS 3015, no issue with this.

But the other anomaly is at 11:30 when G reaches 100% and Predbat switches to Freeze Charge when G briefly starts discharging at full rate.  I'm wondering if this is due to the sequence of commands that are sent to the inverter to change from Charging to Freeze Charge?

Other slightly weird behaviour is from 10:30 on H and 11:30 on both when the SoC fluctuates from 100% and the battery power rises briefly, the battery charges a bit more, then the power stops.  I think this is down to the batteries "settling down" at 100% full, the BMS realising there is a bit more room and letting them charge a bit more.  Don't think this is predbat-related.

If you can look at the switchover from charging to freeze charging though please 

Logfile attached

chalfontchubby, 2026-10-18:
_Posted by Claude on behalf of @chalfontchubby._

@gcoan we're trying to focus the backlog. Inverter balancing and freeze handling have changed a lot since this was opened in March 2024. If you no longer see cross charging on a recent release, could you close this? If something is still wrong, a fresh focused issue with a current debug.yaml would be easier to act on than this long thread.