springfall2008/batpred #831: iBoost on while discharging

dandwhelan, 2024-03-08:
**Describe the bug**
iBoost is set to be enabled while discharge is planned.

**Expected behavior**
Turn off iBoost while Discharge is active

**Predbat version**

7.16.3

**Environment details**

- Inverter and battery setup
HAOS
8.2KW Battery
Car Charger
iBoost using a manual switch

**Screenshots**
[image]

**Log file**

springfall2008, 2024-03-08:
Fix added to 'main' for the next release

gcoan, 2024-03-08:
> iBoost using a manual switch

@dandwhelan out of interest how are you controlling your iboost from the predbat binary sensor? I've heard that there isn't a home assistant integration?

Thanks

dandwhelan, 2024-03-08:
> Fix added to 'main' for the next release

Sorry seems to be the same. 
[image]

dandwhelan, 2024-03-08:
> > iBoost using a manual switch
> 
> @dandwhelan out of interest how are you controlling your iboost from the predbat binary sensor? I've heard that there isn't a home assistant integration?
> 
> Thanks

Trevor mentioned using iboost active state to control the on and off via an automation. Tonight will be my first night seeing if it works

alias: PredBat PVD iBoost
description: "PredBat has triggered a iboost for gas that's cheaper than electric "
trigger:
  - platform: state
    entity_id:
      - binary_sensor.predbat_iboost_active
condition: []
action:
  - choose:
      - conditions:
          - condition: state
            entity_id: binary_sensor.predbat_iboost_active
            state: "on"
        sequence:
          - type: turn_on
            device_id: b769e704fa14dc91fce5bbd6c0a85bab
            entity_id: 4ff92f2d9d4ea02bb33f6786e2a8d979
            domain: switch
      - conditions:
          - condition: state
            entity_id: binary_sensor.predbat_iboost_active
            state: "off"
        sequence:
          - type: turn_off
            device_id: b769e704fa14dc91fce5bbd6c0a85bab
            entity_id: 4ff92f2d9d4ea02bb33f6786e2a8d979
            domain: switch
mode: single

springfall2008, 2024-03-09:
> > Fix added to 'main' for the next release
> 
> 
> 
> Sorry seems to be the same. 
> 
> [image]
> 
> 

It's a cumulative total for the day so I don't see it increasing during discharge ?

gcoan, 2024-03-09:
> Trevor mentioned using iboost active state to control the on and off via an automation. Tonight will be my first night seeing if it works

Thanks for sharing your automation, it looks great although personally I have never used device_id nor numeric entity_id's, I just use the HA entity name in my automations.  I believe device_id ties the automation to the unique device in HA so if you ever swap the device you need to change the automation - probably a low risk issue, but for me just using text entity id's is more readable.

Anyway I was interested in how you physically are controlling the iboost, whether its a smart switch or a HA control for the iboost itself.  On the givenergy community forum someone said that there wasn't any way to remotely control an iboost, and looking on google I could only find an ESPhome based solution.
So was interested to understand how people were actually physically turning the iboost on and off (finger bot?) so I could expand the documentation

dandwhelan, 2024-03-09:
I'm not sure why HA yaml mode turns my entries into the backend numbers.
When I created it with the gui wizard it does use the entity name.

It's a very long story of finding an alternative to iboost. It once used a
complex method of pulses within 3.6 seconds of when a electricity meter
'ticks' over. Now my solar export price is 15p and gas is cheaper I don't
tend to use it.
So, don't have an iboost but Ive turned by a off and in switch that only
pumps out 3kwh. I use the predbat iboost functionality to use electric when
it's cheaper than gas. That's it. Nothing else.
I could have used an automation to control this but predbat gives me the
confidence knowing it's not draining the battery when turned on.

Trevor, I must have misunderstood the plan, I didn't know the plan gives a
total of kwh used in the day. I thought the plan would tell me when it's
going to start the iboost switch at times of the day electric prices are
below 7.5.

springfall2008, 2024-03-16:
>Trevor, I must have misunderstood the plan, I didn't know the plan gives a
total of kwh used in the day. 

Yes its an incrementing total, the up arrow shows when the iBoost is active.

closed 2024-03-16 by springfall2008

gcoan, 2024-03-17:
I've expanded and updated the documentation, making it clearer that it can be used to control any solar diverter, not just an iBoost