springfall2008/batpred #840: Plan is keeping SOC low - not fully charging - behaviour change?

jameslavery, 2024-03-13:
Recently upgraded to 7.16.5
Running GivEnergy hybrid inverter.
On Octopus Agile import and export.

The plan for today is not charging the battery much at all, and keeping the SOC quite low. I'd expect the behaviour I've had for several weeks of nearly fully charging on cheaper import rates, and then using battery during day. I haven't changed any settings.

Do I need to tune my settings? Plan, tariff and forecast graphs below.

[image]
[image]
[image]
[image]

RobinCu, 2024-03-13:
Hi James, the plan looks fine to me.  Solar generation is forecasted to match or exceed the load forecast from 08:30 to 16:00, and the battery is not forecasted to deplete during any high-rate period.  At times when Agile rates are quite flat, it is better to run off the grid for most of the time outside of the peak 16:00 to 19:00 period to avoid incurring losses.  

More info and recommendations if you want to force some charging or de-risk the forecasts are in the FAQ section: "The plan doesn't charge or discharge when I expect it to". https://springfall2008.github.io/batpred/faq/

Side note: Is there a reason you're on the Agile export rather than Octopus Outgoing Fixed at 15p/kWh? If you were on Outgoing Fixed, and Predbat was set to discharge to the grid at 15p/kWh (before losses), it would actually encourage more charging behaviour at low rate periods.

gcoan, 2024-03-13:
I agree with @RobinCu comments, your solar generation is higher than your house load so predbat is allowing the battery to charge from solar rather than grid. At no point in the 24 hours period does your battery get exhausted.

If you want to increase the stored charge "just in case" then you can increase best_soc_keep but it looks fine as it is to me. It's just a good solar day

jameslavery, 2024-03-13:
Good point on the fact that solar (despite being less than recently) is more than house load.

Just needed a sanity check!

Responding to the side note on Agile export. I was comparing with Intelligent Flux, and Agile is better financially than Intelligent Flux. Hadn't considered, recently at least, just going for Outgoing Fixed. I need to think about that.

Having said that, from the beginning of May we're away for several weeks, so need to think about which tariff to be on, and the PredBat settings for that. We'll be coming in to sunnier times, with the house hardly consuming anything.

gcoan, 2024-03-13:
Get a heat pump and that'll cure any problems with solar being > house load !

When you go on holiday just set holiday days to how many days you are away and Predbat will swap to using just yesterday's consumption to predict todays (rather than your days_previous setting).

It'd be worth you being on a better export tariff for the holiday. Maybe even swap to Flux for the period you're away

jameslavery, 2024-03-13:
> Get a heat pump and that'll cure any problems with solar being > house load !
>
> When you go on holiday just set holiday days to how many days you are away and Predbat will swap to using just [...]

Setting holiday days - yup was aware of that.

I agree, I think swapping to Flux could be worth it. As long as I can swap back to Agile on our return.

closed 2025-01-01 by springfall2008