springfall2008/batpred #841: Predbat mode changes on reboot

colinm1, 2024-03-13:
**Describe the bug**
When Home Assistant is restarted Predbat mode changes to Monitor from Control Charge & Discharge

**Expected behavior**
Mode to remain as previously defined

**Predbat version**
7.16.5

**Environment details**

- Inverter and battery setup: Givenergy 3.6 gen 2 hybrid
- Standard HAOS installer or Docker: HAOS
- Anything else?

[appdaemon.log](https://github.com/springfall2008/batpred/files/14589657/appdaemon.log)

danbloor, 2024-03-14:
I found this same issue today

**Predbat version**
7.16.1

**Environment details**

- Inverter and battery setup: GivEnergy AIO
- Standard HAOS installer or Docker: HAOS (seperate AppDaemon not combined)

springfall2008, 2024-03-15:
I might have an idea as to the issue, pushing a fix

colinm1, 2024-03-16:
Still hasn't fixed for me. I'm on 7.16.7 and still rebooting to monitor.

mpartington, 2024-03-22:
Not fixed for me either, although I see a different set of settings change (occasionally) following a reboot.  For me it turns expert mode off and inverter notifications on (hence I know it's done it).  There may be some others as well.  Log attached, this happened shortly before 10:25 on 22-03 (today).

At 10.25, I selected a pre-saved setting via the drop down to restore to my preferences.

[appdaemon.log](https://github.com/springfall2008/batpred/files/14721204/appdaemon.log)

colinm1, 2024-03-22:
> Not fixed for me either (7.16.8), although I see a different set of settings change (occasionally) following a reboot. For me it turns expert mode off and inverter notifications on (hence I know it's done it). There may be some others as well, however I remain in charge/discharge mode. Log attached, this happened shortly before 10:25 on 22-03 (today).
> 
> At 10.25, I selected a pre-saved setting via the drop down to restore to my preferences.
> 
> [appdaemon.log](https://github.com/springfall2008/batpred/files/14721204/appdaemon.log)

It also changes expert mode  for me. I now have an automation to revert the settings if it detects a change.

colinm1, 2024-03-22:
It also changes expert mode for me. I now have an automation to revert the settings if it detects a change.

springfall2008, 2024-11-16:
This should be fixed now, I re-wrote the settings to be saved to a file

closed 2024-11-16 by springfall2008