springfall2008/batpred #842: Notifications spamming

bowen159, 2024-03-13:
**Describe the bug**
HA is sending push notifications every 5 minutes even though the inverter state hasn't changed and battery percentage is the same. This seems to happen when freezing discharging.

**Expected behavior**
I notification should only occur or could be configured to only send if the inverter state changes.

**Predbat version**

v7.16.5

**Environment details**

- Giv Hybrid Gen1 5kwh and 8.2kwh Gen battery
- HAOS VM on Proxmox

Example of the issue
[image]

**Log file**
Can't get the logs just now. Can add later if still required.

springfall2008, 2024-03-15:
Fix on 'main' for testing

gcoan, 2024-03-15:
Thanks, I'm installing it now and will let you know how I get on.

Was seeing multiple repeated messages every 5 minutes this morning for Discharging, Hold Charging and Charging so be good to see this cured.
Idle seems to be the only status that doesn't repeat that I can see.

gcoan, 2024-03-16:
The charging notification text is different, I thought it was showing current soc to target soc but the figures seemed wrong.
[image]
[image]

Wondering this morning if it's target soc for the two inverters which is exciting if it is 

Overnight definitely less notifications and isn't the repeated identical notifications.
[image]

Although could be argued that the notifications with same Hold Charging but different targets could be suppressed and swap between Charging and Hold Charging also candidates for basically being the same.
But overall much better

springfall2008, 2024-03-16:
I've made a new release, I still need to clean up dual inverter notifications however

gcoan, 2024-03-16:
I'll put the new release in then thanks

closed 2024-03-16 by springfall2008