springfall2008/batpred #845: Home Assistant outage when running appdaemon-predbat

rholligan, 2024-03-14:
**Describe the bug**
I’m experiencing some issues with Home Assistant when running appdaemon-predbat add-on (I don’t experience these issues when I stop the add-on). I’m running everything on a Home Assistant Green. 
Predbat starts up without issue, generates a plan and runs for a couple of hours. Then Home Assistant becomes completely unresponsive, UI is not accessible through port 8123 (or Observer port 4357).
Sometimes the system seems to recover, and other times I just perform a hard reset (Predbat start on boot disabled)
The issue seems to affect the whole system, e.g. there is data outage from other entities (heating system) when Home assistant is locked up (see sample chart).
 

I’ve been having a look through predbat.log, and once system comes back online, there is always a few lines similar to these:
2024-03-13 20:31:52.602363 WARNING AppDaemon: Excessive time spent in utility loop: 130529.0ms, 130395.00000000001ms in check_app_updates(), 133.99999999998545ms in other
2024-03-14 10:54:42.494363 WARNING AppDaemon: Excessive time spent in utility loop: 1730222.0ms, 40099.0ms in check_app_updates(), 1690123.0ms in other

I’ve also noticed a few instances of the inverter clock getting out of sync – could potentially be related? I do have 
2024-03-13 18:56:59.905151 INFO pred_bat: Invertor time 2024-03-13 17:11:54+00:00 AppDaemon time 2024-03-13 18:56:59.895295+00:00 difference -105.1 minutes
2024-03-13 18:56:59.915643 INFO pred_bat: WARN: Invertor time is 2024-03-13 17:11:54+00:00 AppDaemon time 2024-03-13 18:56:59.895295+00:00 this is -105.1 minutes skewed, Predbat may not function correctly, please fix this by updating your inverter or fixing AppDaemon time zone
2024-03-13 18:56:59.990789 INFO pred_bat: WARN: Inverter control auto restart trigger: Clock skew >=5 minutes
2024-03-13 18:57:00.010091 INFO pred_bat: Calling restart shell command: rm -rf /homeassistant/GivTCP/*.pkl
2024-03-13 18:57:00.225886 INFO pred_bat: Calling restart service hassio/addon_restart with addon a6a2857d_givtcp
2024-03-13 18:57:15.291712 INFO pred_bat: ERROR: Exception raised Auto-restart triggered
2024-03-13 18:58:35.342165 WARNING AppDaemon: Excessive time spent in utility loop: 5619.0ms, 5544.0ms in check_app_updates(), 75.0ms in other
2024-03-13 19:31:50.754579 INFO AppDaemon: Previous message repeated 507 times

**Expected behavior**
Predbat to run in the background without impacting functionality of Home Assistant

**Predbat version**
appdaemon-predbat  version: 1.0.8
Predbat version 7.16.5

**Environment details**

- Inverter and battery setup: Givenergy 5kW hybrid inverter Gen 3 / 9.5kWh battery Gen 3
- Standard HAOS installer or Docker: standard (Home Assistant Green)
- Anything else?

**Screenshots**
[image]

**Log file**
predbat.log https://pastebin.com/ZQTWhSEk 
Appdaemon.yaml https://pastebin.com/4CxTtwaY
Apps.yaml https://pastebin.com/NUQX3w8k 
Configuratino.yaml https://pastebin.com/BLRw0fHr

springfall2008, 2024-03-14:
Hmm it seems like it locked up, not sure exactly why that might be. Can you try changing to 'threads: 0' to disable multithreading? A value of 1 isn't a good choice, either use 2 or more or leave it disabled.

rholligan, 2024-03-14:
Will give that a try. Threads was set to "auto" before when this issue first occured, I tried changing to "1" but didn't improve things.

rholligan, 2024-03-15:
Has been running successfully overnight, issue appears to be resolved. Thanks for quick response!

closed 2024-03-15 by rholligan