springfall2008/batpred #846: Minimum battery percentage - Solaredge focussed but may apply to other inverters

LupoLoopy, 2024-03-14:
**Is your feature request related to a problem? Please describe.**
To reduce the wear on my SolarEdge battery and to reduce the likelihood of it entering sleep mode, I try to avoid discharging the battery to lower than 3% with my current NodeRed flows.

**Describe the solution you'd like**
I'd like a minimum battery percentage that BatPred will drive the battery down to.  With the SolarEdge module specifically, when the battery gets down to said percentage, I would like the holding behaviour storage control mode to be 'Charge from Solar Panel'. Once the threshold is breached through recharge from solar, 'Maximise Self Consumption' would be a decent setting.

**Describe alternatives you've considered**
I could drive this behaviour with a NodeRed based watchdog using the resources I've already created, but that would entail running two parallel solutions attempting blind control of a shared resource.

RobinCu, 2024-03-14:
Predbat already has input_number.predbat_best_soc_min (in kWh) which should do what you are trying to achieve.  

There is also input_number.predbat_best_soc_keep which is a soft target that the plan will aim to achieve but it will drop below the keep figure if load exceeds forecast.

LupoLoopy, 2024-03-14:
I'll close this since the functionality already exists! Thank you RobinCu

closed 2024-03-15 by springfall2008