springfall2008/batpred #848: Switch charger to off during force discharge slots

rknight20772, 2024-03-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

RobinCu, 2024-03-15:
The best way to do this is increase your export margin in the zappi settings (on the physical device menu, rather than the app).  Set it to something higher than you inverter's max output (5000W). Specific instructions are here: https://support.myenergi.com/hc/en-gb/articles/18468235308433-How-to-stop-menergi-products-diverting-exported-power#h_01GZEF0YMHWH8GCR4R14R52HJE

springfall2008, 2024-03-16:
Predbat has a sensor 'binary_sensor.predbat_discharging' which you can use for an automation trigger to change your Zappi mode

gcoan, 2024-03-18:
> Predbat has a sensor 'binary_sensor.predbat_discharging' which you can use for an automation trigger to change your Zappi mode

I've added details of this to the documentation as it was missing

closed 2025-01-01 by springfall2008