springfall2008/batpred #849: Updating from v7.16.1 to v7.16.5 causes various settings to be changed including setting the mode to Monitor

RonS55, 2024-03-15:
**Describe the bug**
I updated from v7.16.1 to v7.16.5.  I initially updated from the usual HA settings notification as there were also HA OS updates identified there as well.
I then noticed that there were some HACS updates such as Mushroom card etc.  One of these updates was also Predbat.  I updated all including the Predbat one although perhaps I shouldn't have as I had already updated via the HA settings notification.

When HA and Predbat restarted I noticed that the schedule was not showing any charging and then found it was in Monitor mode.  I changed the mode to Control Charge, which is how I normally run.  I then noticed that several of the other switches and values had been changed (possibly to default values).  For example Expert mode was switched off and EV battery values were set on (I don't have an EV so switched these off some time ago).  I then noticed that possibly some of the other values were different.  At this point I went through and updated all of the settings to what I had before. 

**Expected behavior**
Based on previous updates to Predbat, I expected to update and then possibly restart the server (Raspberry Pi4) and then the Predbat functionality would continue working from where it left off before the update.

**Predbat version**

v7.16.5

**Environment details**

- Inverter and battery setup  GivEnergy AC 3.0 inverter and 2 x 5.2kWh batteries.  Solar via a Growatt inverter
- Standard HAOS installer or Docker - Standard HAOS running on a Raspberry Pi4 using a 256GB SSD drive
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Log file from Predbat is attached.  I think the update is identified at 18:02:44 and then you can see that mode is set to Monitor.  You can also see that previously in this log the mode and other values such as Expert mode were different.
[appdaemon.txt](https://github.com/springfall2008/batpred/files/14618120/appdaemon.txt)

springfall2008, 2024-03-15:
Potential fix here: https://github.com/springfall2008/batpred/commit/ae1b528deb12029d850b541de494f9910b9faec1

closed 2024-03-16 by springfall2008

RonS55, 2024-03-17:
Updated to v7.16.7 and settings now seem to be untouched as expected.