springfall2008/batpred #853: Update notifications contain changelog for previous update

ablack89, 2024-03-15:
**Describe the bug**
When an update to predbat is available the notification from Home Assistant contains a changelog, but this changelog is actually the one for the currently installed (out of date) version. The changelog for the new version is only shown in the version information after the update has been completed.

**Expected behavior**
The changelog shown on the update notification would allow the changes in the latest (yet to be installed) version.

**Predbat version**

7.16.5

**Environment details**

- installed using appdaemon-predbat

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

ablack89, 2024-03-17:
I have captured a screenshot of the update notification before the upgrade to 7.16.7:
[image]
And immediately after installing the update:
[image]

springfall2008, 2024-03-17:
Fix now on 'main' will be released soon

springfall2008, 2024-03-17:
https://github.com/springfall2008/batpred/releases/tag/v7.16.8

ablack89, 2024-03-17:
Good work, thanks.

closed 2024-03-17 by ablack89