springfall2008/batpred #857: Random errors and warnings

SwiftRR, 2024-03-16:
**Describe the bug**
In Predbat, I am receiving strange status reports which then disappear. See the screen shots.
Are these anything to worry about?
One of the warnings refers to REST. Do I need this set up in my apps.yaml file. Currently it is:
  givtcp_rest:
   - http://192.168.1.239:6345
   - http://192.168.1.239:6346
(as in your video)
I have also attached my apps yaml file
[apps-2.yaml.txt](https://github.com/springfall2008/batpred/files/14624106/apps-2.yaml.txt)
**Expected behavior**
No warnings of exceptions

**Predbat version**
v7.16.7
xxxx

**Environment details**

- Inverter and battery setup
- Inverter and battery setup: Givenergy AC 3.0 coupled inverter daily chaining to two 8.2 kWh batteries
Standard HAOS installer or Docker: Raspberry Pi 4.0 operating HA connected to a 500 Gb SSD drive (Samsung T7)
I access Predbat on various Mac devices, iMac, iPhone and iPad.

**Screenshots**
Snapshots of recents status entries:
[image]
[image]

**Log file**
Log files covering today (16-03-2024)
[predbat.log.1.txt](https://github.com/springfall2008/batpred/files/14624109/predbat.log.1.txt)
[predbat.log.2.txt](https://github.com/springfall2008/batpred/files/14624110/predbat.log.2.txt)
[predbat.log.3 2.txt](https://github.com/springfall2008/batpred/files/14624112/predbat.log.3.2.txt)
[predbat-4.log.txt](https://github.com/springfall2008/batpred/files/14624121/predbat-4.log.txt)

springfall2008, 2024-03-16:
It seems like you have some sort of intermittent connection issue or your server is running quite slowly.

What polling time did you set with GivTCP?

SwiftRR, 2024-03-16:
How do I find the polling time?

I have looked at the givtcp device settings in HA under MQTT.
See logbook entries below which suggests that the inverter time is being changed every few seconds. 
[image]

Rob

gcoan, 2024-03-16:
you need to look at the GivTCP add on, and then the configuration.  Its the refresh frequency

SwiftRR, 2024-03-16:
I think I have found it. See screenshot.
[image]

I have since changed the loop timer to 5 seconds as this is stated as the default but I have not changed the Queue retries.
I couldn't find an entry for 'refresh frequency'.
I remember that I set up this add-on using Speak to the Geek video.
I am running MQTT v6.3.1. There does appear to be an update v6.4.0 but I am always anxious that HA updates could break things.
Yesterday I also restarted the Inverter to defaults.

gcoan, 2024-03-16:
5 might be too frequent as this is how often the inverter is being polled by GivTCP.  If its too frequent then the inverter spends all its time responding to queries and doesn't have any 'bandwidth' to accept commands you send.  You're also more likely to get timeouts from the portal and app.

I have mine set to 20 seconds.

The REST failure is predbat failing to talk to GivTCP (which talks to the inverter).  So as Trefor says could be a symptom that whatever you are running GivTCP and Predbat on is overloaded.

SwiftRR, 2024-03-16:
OK. I have changed the Loop timer to 20 seconds.
Should I leave the Queue retries at 2 ?

I have looked again at my GE Power graph screenshot (attached again below). The behaviour mirrors what I would expect for a calibration starting around 8 am and completing around 6:30 pm,
  
[image]

gcoan, 2024-03-17:
I have my queue retries set to 2 and it seems to work OK

Yes that does look like a battery calibration activity, full discharge and charge

SwiftRR, 2024-03-17:
I checked status this morning (17th March 2024) and get the following from early hours:
[image]

My predbat log for that time shows this. See attached log.
[predbat.log-2.txt](https://github.com/springfall2008/batpred/files/14626827/predbat.log-2.txt)
In the log, connection to REST often requires a 2nd attempt. Should I increase the retries in the MQTT/givtcp configuration file?

Also ...... I have checked the log file for givtcp and I often see an error: e.g.
2024-03-17 08:06:54,889 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 failed: (<class 'AssertionError'>, AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response'), <traceback object at 0x7f9a451000>)
In my current predbat log at the same time, there is no error or warning.

**Other clues**
In the MQTT/givtcp configuration file, I have to select the inverter switch for 'Use old firmware'. My inverter is GE 3.0 AC coupled. The firmware is old (D0.535-A0.535) but there is no option for me to upgrade it (several updates shown in GE KB). Forums suggest that GE is drip feeding updates to users.
I am on Battery firmware 3015, which I upgraded to (from 3012) on 5th Jan 2024.  

The predbat log often includes a warning: e.g.
2024-03-17 08:07:54.470519 WARNING AppDaemon: Excessive time spent in callback 'run_time_loop() in pred_bat', Thread 'thread.thread-0' - now complete after 174.469529 seconds (limit=120.0)

predbat, support and documentation are all brilliant and most of the time predbat works flawlessly.
I just want to make 'most of the time' into 'all of the time'.

Thank you for your continued support. I will be onto GE early this week about my random calibrations with no GE notification.

Rob

SwiftRR, 2024-04-01:
Hi, predbat has been working just fine for me but I am still receiving random warnings in status reports. there seems to be no pattern – I can go 2 days with no warns and then received several on the same day. I sometimes get the same warning 3 times (for the same time!)

I have attached a snapshot from earlier today, together with my predbat.log that coincides with some of today's warnings.
[image]
[predbat.log-2.txt](https://github.com/springfall2008/batpred/files/14828413/predbat.log-2.txt)

I have carried out a lot of checking of my wifi network including swapping out a potentially flaky network switch. I have a bthomehub 2 router with 3 black disks which claim to set up a mesh network. Each disk is ethernet backhauled and my GE inverter is claimed to have an excellent connection. I am thinking of getting a 3rd party wireless setup and possibly ditching my bt router (I am on 500 Mb fibre to premises).
My GE inverter is sited in the same area of my SolarEdge PV inverter and this has never dropped of my network (yet!)

Do you think these status warning are just that and 'just one of those things' that happens with givtcp? Perhaps I am getting neurotic about the warnings?

Thanks 

Rob

gcoan, 2024-04-01:
Looking in my predbat statii, I had the following rest errors:

- inverter 0 this morning at 3:24, worked fine at 3:25
- inverter 1 yesterday 10:31, fine at 10:36
- inverter 1 3 days ago 11:13, fine at 11:15

I have two inverters so double the chance of errors as you.

Maybe getting 1 a day on 1 inverter but it usually clears soon afterwards

SwiftRR, 2024-04-01:
Thanks. 
I will keep my fingers crossed and live with the warnings. I will try not to check my status reports as often! 
As you can see from my status warnings, whenever I have experienced a warning, the next predbat commands is usually accepted, which is similar for you. 
I think the issue is more likely to be with MQQT and givtcp than with predbat (or my 'ancient' AC coupled inverter). 

Rob

springfall2008, 2024-11-16:
Closing this old ticket, please open a new one if you have an up to date question

closed 2024-11-16 by springfall2008