springfall2008/batpred #858: Constant hold charge at 100%

Philhews, 2024-03-17:
**Describe the bug**
After charging to 100 the current clot is always set to hold charge 100 even when the grid cost is higher than the export of excess solar (even to the point of the first expensive agile slot) 

**Expected behavior**
When grid import cost goes above export price the system should be idle to allow battery use and refill rather than forcing grid use at expensive rates. 

**Predbat version**

7.16.7

**Environment details**

- hybrid 3.6 gen 2, 9.5kwh battery
- Standard HAOS 
- Anything else?

**Screenshots**
[image]

**Log file**
[appdaemon.log](https://github.com/springfall2008/batpred/files/14626949/appdaemon.log)

springfall2008, 2024-03-17:
I think this is due to having "metric_min_improvement(-0.4 p)" meaning you will charge now even if it costs 0.4p extra.

Can you try setting it back to 0?

Philhews, 2024-03-17:
Don't think that's helped. 
[image]

Philhews, 2024-03-17:
Log file from it too
[appdaemon.log](https://github.com/springfall2008/batpred/files/14627080/appdaemon.log)

Philhews, 2024-03-17:
Having forced an idle and used some battery it seems it's happy to leave it on idle when below 100
[image]

springfall2008, 2024-03-17:
Can you try changing 'metric_battery_value_scaling' to 1.0?

springfall2008, 2024-03-17:
Also I have a new release coming that might help with this too..

springfall2008, 2024-03-17:
Please try this one: https://github.com/springfall2008/batpred/releases/tag/v7.16.8

Philhews, 2024-03-18:
Battery value scaling at 1, metric min improvement set to 0 and updated to .8 seems to look more sensible as an ahead plan, will leave it and keep an eye on what it does during the day. 
[image]

Philhews, 2024-03-18:
Having pulled in nord pool rates it went back to hold charge on an expensive slot. (10.10 plan). Changing metric min improvement to -0.2 removed the charges except on the current slot which is over 16p import. 
[image]
[image]
[appdaemon.log](https://github.com/springfall2008/batpred/files/14634687/appdaemon.log)

Philhews, 2024-03-18:
Initially the hold charge went away at 4pm,but at 5 past it calculated and put it back in. This results in anything over solar pulling at 26p which can't be the best plan? 
[image]
[appdaemon.log](https://github.com/springfall2008/batpred/files/14639281/appdaemon.log)

springfall2008, 2024-11-16:
Closing this old ticket, please open a new one if you have an up to date question

closed 2024-11-16 by springfall2008