springfall2008/batpred #860: Load "prediction" too high with anything other than "days_previous" set to 7

pacemaker82, 2024-03-17:
**Describe the bug**
I have been using `days_previous` set to 7 since install. I have IOG and charge my car every night. I have noticed that my load prediction during the night doesnt seem to include the car charging load. Below is the prediction with `days_previous` set to "7"

[image]

So today I've been experimenting using multiple `days_previous` and I get really high loads during the IOG charge window. Here I am using `days_previous` with just "2"

[image]

Regardless of what I use other than "7", I get really high loads during the IOG charge time. Is something off here? I wanted to set my `days_previous` to 2,3,4,5,6,7,8 to average out my load over the week, which works great for all the times except the IOG cheap slot timing where the load prediction goes really high.

Its almost like the car charge isnt included when I use "7" but is included if I use any other value for `days_previous`

Right now there is no planned car charge for tonight. Not sure if that matters (car column is hidden)

**Expected behavior**
I am not sure - clarity on if the load includes the car load or not, and therefore why using "7" shows a low load, but any other value shows a high load.

**Predbat version**

7.16.7

**Environment details**

- Inverter and battery setup: AC-3, 9.5kWh batt GE
- Standard HAOS installer or Docker: HAOS
- Anything else?

**Screenshots**
above

springfall2008, 2024-03-17:
Do you have car_charging_energy set correcting in apps.yaml?

pacemaker82, 2024-03-17:
Ahhhh, thank you that was it. The sensor I had set was related to tesla integration which I uninstalled. Changed it to the new sensor and working great now!

This is for `days_previous` 2,3,4,5,6,7,8
[image]

closed 2024-03-17 by pacemaker82