springfall2008/batpred #863: Battery charge mode

mjs7uk, 2024-03-17:
Hi first thanks for a great tool and all of your efforts in producing and supporting this.

Would it be possible to get switch.predbat_plan_debug available in non expert mode as it really helps understand why the plan might be the way it is.

Second thing and sorry if this operator error, given the current rates on flux with losses it does not make sense to do a full charge over night, really it looks like you you should do a enough of a charge to get to the point where the solar produces enough to cover house load and start charging the battery, and then only do an export to get to the next charge point

``

 
[image]

gcoan, 2024-03-17:
Have a look at this faq on tuning the predbat plan https://springfall2008.github.io/batpred/faq/#the-plan-doesnt-charge-or-discharge-when-i-expect-it-to

It's likely you might need to tweak your loss percentages. What are they currently set to ?

mjs7uk, 2024-03-17:
Hi currently set to 5,5,5. I have enphase micro inverters that output ac. I am running in non expert mode.

All the others seam to match the suggested settings
[image]

gcoan, 2024-03-17:
5,5,5 is reasonable and should model correctly whether Flux is worthwhile to charge overnight or not.

Is it possible to copy the predbat plan so we can see the detailed numbers.

I *think* what I am seeing is:
- an overnight charge to full battery
- then the SoC dropping down (using the cheaper stored electricity) in the morning to about half full
- solar generation then starts to reverse the red line, filling the battery up to 100% just before the Flux peak period
- exporting during the flux peak period to about half-full
- house load then emptying the remainder of the battery to end of the night

If I've interpreted the graph correctly then this is a reasonable plan.  You look to have high morning load and only just enough solar to then recoup the battery up to 100% before the evening peak

mjs7uk, 2024-03-17:
Hi thanks for looking at this, here is the plan (3 images for each).have switched to expert mode to.get the losses, the first plan (first 3 images) is with no best max soc set the second I set it to 7 just to see if my thinking that it did not need a full charge was roughly right it looks like they come out pretty close 
[image]
[image]
[image]
[image]
[image]
[image]

gcoan, 2024-03-17:
Looking at the first plan, it is much as I thought it was.  There isn't that much peak rate discharge, just a bit of excess to get your battery to a level that will take you through to the next overnight charge.

There is potential for the overnight charge to not be as big as it is as there's 4 hours of export in the afternoon when the battery is at 100%.  But conversely the solar might not be as good as predicted and your house load could be higher, so some contingency in the plan isn't an absolutely bad thing.

springfall2008, 2024-11-16:
Closing this old ticket, please open a new one if you have an up to date question

closed 2024-11-16 by springfall2008