springfall2008/batpred #867: TypeError: argument of type 'NoneType' is not iterable

onetoomany, 2024-03-18:
From latest version:
 v7.16.8

```
2024-03-18 10:55:02.317144 WARNING pred_bat: ------------------------------------------------------------
2024-03-18 11:00:01.966605 WARNING pred_bat: ------------------------------------------------------------
2024-03-18 11:00:01.966914 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-03-18 11:00:01.967169 WARNING pred_bat: Worker Ags: {'id': '2fc90665927d42ff87d845ca949acd9c', 'name': 'pred_bat', 'objectid': '9166e474f15c41a48e746bc20b80f3ca', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7f3d6ebd9110>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-03-18 11:00:01.967434 WARNING pred_bat: ------------------------------------------------------------
2024-03-18 11:00:01.969599 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 12671, in run_time_loop
    raise e
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 12667, in run_time_loop
    self.update_pred(scheduled=True)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 11664, in update_pred
    status, status_extra = self.execute_plan()
                           ^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 10622, in execute_plan
    inverter.adjust_battery_target(self.charge_limit_percent_best[0], isCharging)
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 2984, in adjust_battery_target
    self.mimic_target_soc(soc)
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 2775, in mimic_target_soc
    self.set_current_from_power("charge", (0))  # Set charge current to zero (i.e hold SOC)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 3420, in set_current_from_power
    entity = self.base.get_entity(self.base.get_arg(f"timed_{direction}_current", indirect=False, index=self.id))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/adapi.py", line 3188, in get_entity
    self._check_entity(namespace, entity)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 231, in inner_sync_wrapper
    f = run_coroutine_threadsafe(self, coro(self, *args, **kwargs))
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 313, in run_coroutine_threadsafe
    result = future.result(self.AD.internal_function_timeout)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 456, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/usr/lib/python3.11/site-packages/appdaemon/adapi.py", line 588, in _check_entity
    if "." not in entity:
       ^^^^^^^^^^^^^^^^^
TypeError: argument of type 'NoneType' is not iterable
```

stevebuk1, 2024-03-19:
Apologies - this was an error introduced in #843. Fix required is [here](https://github.com/springfall2008/batpred/pull/843#issuecomment-2008050706)

springfall2008, 2024-03-28:
https://github.com/springfall2008/batpred/releases/tag/v7.16.9

closed 2024-04-02 by onetoomany