springfall2008/batpred #868: V7.16.8 Plans to charge battery fully in cheap slot even though plenty of PV is expected.

SJB1235, 2024-03-18:
**Describe the bug**
I'm on Octopus Flux. Since v7.16.5 Predbat has been generating plans with grid charging of the battery being reduced to take into account predicted PV generation. In v7.16.8 Predbat is back to planning to grid charge the battery as much as possible in the cheap slot.

**Expected behavior**
I would expect that grid charging be reduced to take into account PV generation.

**Predbat version**
v7.16.8 

**Environment details**
- Gen 1 3.6 Hybrid Inverter and 8.2kWh Battery
- HAOS 12.1

**Screenshots**
v7.16.7
[image]
[image]

v7.15.8
[image]
[image]

Settings:
switch.predbat_expert_mode=on
switch.predbat_active=off
input_number.predbat_pv_metric10_weight=0.75
input_number.predbat_pv_scaling=1.0
input_number.predbat_load_scaling=1.1
input_number.predbat_load_scaling10=1.0
input_number.predbat_load_scaling_saving=1.0
input_number.predbat_battery_rate_max_scaling=0.98
input_number.predbat_battery_rate_max_scaling_discharge=1.0
input_number.predbat_battery_loss=0.05
input_number.predbat_battery_loss_discharge=0.05
input_number.predbat_inverter_loss=0.04
switch.predbat_inverter_hybrid=on
switch.predbat_inverter_soc_reset=off
switch.predbat_battery_capacity_nominal=off
input_number.predbat_car_charging_energy_scale=1.0
input_number.predbat_car_charging_threshold=6.0
input_number.predbat_car_charging_rate=7.4
input_number.predbat_car_charging_loss=0.08
input_number.predbat_best_soc_min=0.0
input_number.predbat_best_soc_max=0.0
input_number.predbat_best_soc_keep=0.33
input_number.predbat_metric_min_improvement=0.0
input_number.predbat_metric_min_improvement_discharge=5.0
input_number.predbat_metric_battery_cycle=0.0
input_number.predbat_metric_battery_value_scaling=1.0
input_number.predbat_metric_future_rate_offset_import=0.0
input_number.predbat_metric_future_rate_offset_export=0.0
input_number.predbat_metric_inday_adjust_damping=0.95
switch.predbat_metric_cloud_enable=off
input_number.predbat_set_reserve_min=4.0
input_number.predbat_rate_low_threshold=0.0
input_number.predbat_rate_high_threshold=0.0
switch.predbat_car_charging_hold=on
switch.predbat_car_charging_manual_soc=off
switch.predbat_octopus_intelligent_charging=on
switch.predbat_octopus_intelligent_ignore_unplugged=off
switch.predbat_car_charging_plan_smart=off
switch.predbat_car_charging_from_battery=off
switch.predbat_calculate_discharge_oncharge=on
switch.predbat_calculate_fast_plan=off
switch.predbat_calculate_second_pass=off
switch.predbat_calculate_tweak_plan=off
switch.predbat_calculate_regions=on
switch.predbat_calculate_inday_adjustment=on
input_number.predbat_calculate_plan_every=10.0
switch.predbat_combine_charge_slots=on
switch.predbat_combine_discharge_slots=off
switch.predbat_set_status_notify=on
switch.predbat_set_inverter_notify=off
switch.predbat_set_charge_freeze=off
switch.predbat_set_charge_low_power=off
switch.predbat_set_reserve_enable=on
switch.predbat_set_discharge_freeze_only=off
switch.predbat_set_discharge_during_charge=on
switch.predbat_set_read_only=off
switch.predbat_balance_inverters_enable=off
switch.predbat_debug_enable=off
switch.predbat_auto_update=off
switch.predbat_load_filter_modal=on
switch.predbat_iboost_enable=off
switch.predbat_iboost_solar=on
switch.predbat_iboost_gas=off
switch.predbat_iboost_charging=off
input_number.predbat_holiday_days_left=0.0
input_number.predbat_forecast_plan_hours=24.0
switch.predbat_plan_debug=on

**Log file**
[appdaemon.log](https://github.com/springfall2008/batpred/files/14635343/appdaemon.log)

springfall2008, 2024-03-24:
Given your export rate is very close to your cheap import rate and your peak export rate is higher I think it makes sense to charge to 100% and account for good vs bad solar days and low vs higher usage days?

SJB1235, 2024-03-24:
The v7.15.8 plan was more expensive for the first day which didn't seem very good to me. The worst case prediction for the next day pretty much matched the usual load so it seemed strange to do a full top up when only a small one was needed.
I like to not import anything (or as little as possible) if solar can supply power the next day. 

I'll look at reducing the number of charging slots available when the predicted PV is high enough to justify it.  Something I was planning on doing for the summer as I like to mitigate the clipping on my system by starting the day on almost empty battery and charging at a lower rate.

Thanks for looking at this Trefor.

gcoan, 2024-03-24:
> Given your export rate is very close to your cheap import rate and your peak export rate is higher I think it makes sense to charge to 100% and account for good vs bad solar days and low vs higher usage days?

I understand what you are saying Trefor but the Flux overnight import rate (after losses) is 1.5p more expensive than the daytime export rate so my expectation has been that it's not cost effective to charge overnight unless it's predicted to be a poor solar day that won't fill the battery the next day.

I'm wondering how the plan evolved when predbat can "see" the whole of the next day pv figures. Does the overnight charge period now reduce so the battery ends on 100% just a short while before the peak?

closed 2025-01-01 by springfall2008