springfall2008/batpred #869: Status - Calibration

TheSargey, 2024-03-18:
**Describe the bug**
Batpred went into "calibration" status and the battery switched to full discharge. However the inverter has not given any notification of the event. 

Is this normal behaviour? I'm assuming batpred can't initiate a calibration and that the inverter is doing this (I have a Givenergy 3.6 gen 1 inverter). 

Just wanted some clarification from those in the know as I've not had a calibration event in months on my batteries/inverter and when it did the last one the inverter notified me.  I've only been using batpred for a few days and didn't know whether this is something that cna be instigated via the app. 

Thanks as always for any guidance.

Sarge

**Predbat version**
7.16.8

gcoan, 2024-03-18:
Correct, Predbat can't initiate a calibration.  Its the inverter that has initiated a calibration which is a full discharge then a full recharge, and Predbat now notices this and pauses operation.

The functionality for Predbat noticing that the inverter is calibrating the battery is relatively new so it may be that there have been previous calibrations that you were unaware of.

Anyway, Predbat is not initiating, only identifying that the calibration is running. Its a question for GivEnergy as to why this has happened.

@swiftrr had a similar event on #856 so maybe these calibrations happen and up to now we've not known?

TheSargey, 2024-03-18:
Thanks @gcoan - it seems to have coincided with a "BMS over voltage" which I've now received notification of.  It was only that I was working on my dashboard that I noticed I was exporting over 1kW back to the grid (which, being that I don't have an export tariff suprised me!) and obviously the BatPlan was set to charge and not discharge - hence I noticed this anomaly.  Also, the solar stopped providing energy (i.e. it went from generating 1kW at the time to zero) so obviously something went awry but as there wasn't any indication from the inverter to say so I was just trying to figure out what was happening.

However, whilst I was looking at the inverter I found there was a battery firmware update ready (I think it's been sat there for a while waiting for me to initiate it) so I did this at the same time which meant the calibration wasn't just done "for fun" and I managed to upgrade my firmware a couple of revisions whilst it sorted itself out.  All good now.

Happy pre-bat-dicting!

closed 2024-03-18 by TheSargey

gcoan, 2024-03-18:
No problem.  It is likely that the BMS over voltage was the triggering event.  Annoying when it happens especially if you don't have an export tariff, and then have to import in peak rate.

I had a calibration done recently but initiated by the installer due to a new battery being fitted.  One thing I did find was that I could put predbat into read-only mode and drop the discharge rate down (e.g. to the level of your house demand),  and then keep the charge rate at zero until the import rates reduced after the peak.  So I did the calibration but I did it in a cheaper way.

You have to put predbat into read only because by default it sets charge and discharge rate to maximum for the calibration