springfall2008/batpred #871: Add functionality for Solax X1 AC Inverter

punkymuzzle, 2024-03-18:
Hi
Absolutely loving the work that everyone has put in here, it's really appreciated.
I have a Solax X1 AC inverter with a Waveshare Modbus attached, and have access to all of the entities for the inverter.
I have installed Predbat and have successfully managed to get it fully monitoring and predicting etc, which is awesome.
The problem is that I can't get it to charge / discharge the battery as per the plan, even though I've enabled charge / discharge in predbat.
I'm assuming that the issue is predbat knowing the exact commands to send to my inverter, so was wondering if anyone could help in solving this with me?
Happy to provide whatever information I can to help get this sorted, as I know others are in the same situation as me.

Many thanks in advance

springfall2008, 2024-03-28:
Please see the thread here, this maybe what you need: https://github.com/springfall2008/batpred/issues/259

There is a solax template now on 'main' together with the latest fix

punkymuzzle, 2024-03-28:
Thanks for this.  However,this is a G4 version (later version) whilst mine
is the G1 version (X1AC).  As I'm finding out via other routes, the
entities are different between then, and my inverter doesn't support some
of the features from the G4 (such as force discharge)

punkymuzzle, 2024-03-29:
Here you go:

punkymuzzle, 2024-04-01:
> More than happy to help where I can.
> It installs and monitors OK, but won't drive the Inverter.
> I'll send some screenshot later.

Do you want me to work with you to prototype this for a SolaX X1 Ac
Inverter?
As I said before, it seems to plot out the charging table etc, but doesn't
actually charge the Inverter when I take it off Read only.
I can supply a full list of entities extracted from the Inverter via Modbus
if that would help?
Many thanks

punkymuzzle, 2024-04-01:
Failing that, as a fallback I can setup some automations to charge my battery based on the table as it's generated?   I know this isn't exactly driving the inverter via predbat, but it would be a tactical fix until the inverter is supported?
.
This would take some quite advanced yaml coding (which is probably beyond me at the moment) but if this is the best way forward I will look to do that.

gcoan, 2024-04-01:
If you can work out the automation command sequences to control the charging it shouldn't be much more to have this integrated inside predbat

punkymuzzle, 2024-04-01:
Yeah I'd love that, but not really sure where to begin.
If I knew what to look for, times, charge, duration etc I'd give it a try.
But all I have is visibility to a charging plan and nothing else.

punkymuzzle, 2024-05-29:
Hi
Is there any progress on this request please?
I have a X1 AC Inverter ready to go and test predbat to see if we can get it to work.
I already have it working with another automation (pv_opt) but also wanted to get it working with predbat so others can also benefit.

Please let me know if you need any additional information from me.
Thanks

chalfontchubby, 2026-09-07:
Hi. Can I ask if this ever landed? Ticket is quite old with no recent comments. 

If it still needs to exist I'll keep it open, otherwise I'll close.

punkymuzzle, 2026-09-07:
Hi
Nah this can be closed thanks.
It never really took off so I used PV-Opt instead which worked really well for me.
Now on foxess inverter on IOG so don't really need 30 minute charge / discharge slots etc.

Thanks

closed 2026-09-07 by gcoan