springfall2008/batpred #874: BatPlan looking to charge for the majority of the day

TheSargey, 2024-03-18:
**Describe the bug**
PredBat is looking to constantly charge the battery throughout the day and only set to idle at 4pm when the expensive rates of import on Agile kick in.

**Expected behavior**
Normally PredBat will balance import with idle to use the battery through the day but ultimately ensure that a full charge has occured by 4pm.  It's unusual to see PredBat charging constantly through the day.  I haven't adjusted settings so I'm unsure how to decipher why it has decided to do this.

I'm wondering what specific variables should I be looking to amend to get a better plan - I've been through the guides on customisation and configuration and seem to be matching what is proposed as best for my setup (Agile import no export).

**Predbat version**

v7.16.8  (tried switching back to 7.16.5 to see if it made any difference)

**Environment details**

- GE 3.6 Gen 1 inverter -- 2 x 5kW batteries
- Standard HAOS installer 

**Screenshots**
[image]

Thanks as always for your help, really appreciated.

Sarge

gcoan, 2024-03-19:
Looking at your plan, a few things stand out:
- you don't have a lot of solar generation compared to your house load, so you are not going to get a lot of solar generation
- the agile rates tomorrow are pretty constant so there is not a lot of benefit of predbat charging up and then letting the battery discharge

Depending on what you have your battery and inverter losses set to, you will lose 10-20% overall on charging and discharging the battery,  so in effect you need that kind of difference in the import rates during a time period to make it worthwhile predbat to let your battery discharge.

My own plan for tomorrow is a very similar shape:
[image]

I think its just a function of the agile rates brought about by the relatively constant agile rates. On a day with more rate variation you'll see a different pattern of battery use

TheSargey, 2024-03-19:
Hi @gcoan, thanks as always for your speedy responses.

I see, makes sense I suppose - I've not really fiddled with the input_number.predbat_inverter_loss, input_number.predbat_battery_loss and input_number.predbat_battery_loss_discharge percentages so I'd assume if I did so this would force PredBat to calculate differently in this scenario.

I think I'm just having a hangover from using Octopus R&D to set charge, which were relatively static timewise but would pick the best slots during a given period to charge.

Basically, I just need to stop micromanaging and let BatPred do it's thing!

Again, thanks for your response, really helpful as always.

Best,

Sarge

closed 2024-03-19 by TheSargey

gcoan, 2024-03-19:
no problem. I saw the issue and thought I could help

People on the givenergy forum have tested conversion losses and seem to end up with answers in the range of 10-20% so as a rule of thumb, input_number.predbat_inverter_loss, input_number.predbat_battery_loss and input_number.predbat_battery_loss_discharge percentages should be set to something in that range, and 15% overall is a reasonable midpoint.  

Mine are 4%, 5.5% and 5% respectively for example. There was a reason why I put charging up to 5.5%, can't remember what it was but there was some (not much) science behind it.

yes it is easy to want to micromanage predbat.  For diminishing pence of benefits usually !