springfall2008/batpred #879: Failed to update SetReserve.

rsarwar87, 2024-03-21:
**Describe the bug**
Changed to Warn - Inverter 0 REST failed to setReserve to 100 got 4
message pops up.

**Expected behavior**
no warning message

**Predbat version**
latest 7.16.8

**Environment details**
docker 

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?

Fix:
uncomment sleep(10) on line 3784

RobinCu, 2024-03-21:
See the recommended fix here: https://github.com/springfall2008/batpred/issues/878

springfall2008, 2024-03-23:
For reference in apps.yaml add:

```
  # Workaround to limit the maximum reserve setting, some inverters won't allow 100% to be set
  # Comment out if your inverter allows 100%
  inverter_reserve_max : 98
```

closed 2024-03-23 by springfall2008