springfall2008/batpred #881: Wrong effective import rate with negative pricing (low priority)

gcoan, 2024-03-22:
**Describe the bug**
In Expert mode Predbat calculates the effective price after battery losses.
Tonight and tomorrow morning there's predicted to be a lot of negative import priced slots

[image]

Looks like the way Predbat is calculating the effective price doesn't take account of the sign of the pricing because the effective import price after losses is being calculated to be **more negative** than the actual import price which doesn't make sense

**Expected behavior**
Effective price after losses on negative priced slots should be higher than than the actual import price, not lower

**Predbat version**
7.16.8
HAOS

RobinCu, 2024-03-22:
I understand the current calculation as correct.  When my battery is charged at 3.6kW, the import is 3.96kW to account for losses. 

So, based on your plan, if the battery is charged with 1.8kWh of energy (3.6kW for 30 minutes) on Sat 11:00 to 11:30  at -2.47p/kWh, the energy cost in the battery drops by 1.8kWh x -2.47p/kWh = 4.446p.  

The import from the grid to account for 5% inverter + 5% battery losses would need to be 1.98kWh, so the billing gain would be 1.96kWh x -2.72p/kWh = 4.89p.  The rate is predicted to be -2.47p, but the billing cost (or gain if -ve) after losses is -2.72p.

gcoan, 2024-03-22:
I think you're right, it just didn't look "logical"

If I take stupidly wrong figures, import rate of -5p and 50% loss on import, to import and store 1kW of energy I will need to actually import 2kW.

In a particular slot I can import at my max inverter rate. Let's say that's 3kW/hour or 1/.5kW a half hour to keep the maths easy.

So I import 1.5kWh which "costs" -7.5p.

I only store 0.75kWh after losses so each kWh has an effective price of -7.5/0.75 = -10p.

Think I might need a wet tea towel over my head to think about this ...

RobinCu, 2024-03-22:
> Think I might need a wet tea towel over my head to think about this ...

Agreed, it certainly took some concentration on my part to figure out if my calculations were correct! It seems counterintuitive, but losses are a good thing when prices are negative.

closed 2024-03-22 by gcoan