springfall2008/batpred #882: After discharge slots, sometimes when predbat reverts to "idle" mode the discharge power is very low

bigsmelly, 2024-03-22:
**Describe the bug**

Seen occasionally - after a predbat discharge slot, and the status reverts back to idle, I would expect eco mode on the inverter and the house load to be supplied from the battery.

However, I have twice seen it where immediately after a discharge, the battery then only supplies about 5w to the house. 

It would seem that the battery discharge power is not being reset when we revert to idle, or at least it's not happend before we have gone to an idle state

Using the givenergy app to reset the charge power kicks it back into life, and the battery starts supplying the house in this idle state

**Predbat version**
7.16.5

**Environment details**

Gen1 5Kw + 9.5 battery (old firmware!)
HAOS on a dual core intel nuc

**Screenshots**

Apologies, didn't take a screenshot, and sadly i don't have a reliable timing.

I will collect a

**Log file

attaching predbat log, it was around 1845 to 1900ish 

Also in givtcp, it looks like there is only a single charge rate. Is this correct ? 

2024-03-22 18:33:40,264 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Eco
2024-03-22 18:33:41,693 - Inv1 - write       -  [INFO    ] - Setting dynamic mode was a success
2024-03-22 18:33:43,865 - Inv1 - write       -  [INFO    ] - Setting shallow charge 35 was a success
2024-03-22 18:33:50,638 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 18:50 - 19:01
2024-03-22 18:33:57,906 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-03-22 18:34:04,662 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 2600 (50)
2024-03-22 18:34:10,795 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
2024-03-22 18:34:17,620 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2024-03-22 18:34:23,657 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
2024-03-22 18:39:24,842 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 18:45 - 19:01
2024-03-22 18:39:33,368 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-03-22 18:39:39,979 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 18:45 - 19:01
2024-03-22 18:39:47,546 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-03-22 18:45:06,244 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 18:40 - 19:01
2024-03-22 18:45:12,913 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-03-22 18:45:34,118 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Timed Export
2024-03-22 18:45:39,658 - Inv1 - write       -  [INFO    ] - Setting export mode was a success
2024-03-22 18:45:42,161 - Inv1 - write       -  [INFO    ] - Enabling Discharge was a success
2024-03-22 18:45:49,906 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 26
2024-03-22 18:45:53,699 - Inv1 - write       -  [INFO    ] - Setting shallow charge 26 failed: (<class 'AssertionError'>, AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response'), <traceback object at 0x7f2d15730980>)
2024-03-22 18:46:00,429 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2024-03-22 18:46:06,547 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2024-03-22 18:51:01,477 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Eco
2024-03-22 18:51:03,034 - Inv1 - write       -  [INFO    ] - Setting dynamic mode was a success
2024-03-22 18:51:05,197 - Inv1 - write       -  [INFO    ] - Setting shallow charge 26 was a success
2024-03-22 18:51:12,121 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 18:50 - 19:01
2024-03-22 18:51:19,347 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-03-22 18:51:25,707 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 18:50 - 19:01
2024-03-22 18:51:33,060 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-03-22 18:51:39,813 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 2600 (50)
2024-03-22 18:51:46,017 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
2024-03-22 18:51:52,800 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2024-03-22 18:51:58,872 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
2024-03-22 18:52:21,149 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Timed Export

[appdaemon.log](https://github.com/springfall2008/batpred/files/14727970/appdaemon.log)

gcoan, 2024-03-22:
Have you got the bit of givtcp logfile immediately after this, showing what happened around 19:00 when the discharge slot finished and Predbat was setting the inverter back to Eco.

Your question on charge rate, if I understand correctly, Predbat pretty much only uses full charge/discharge rate (2600 for a Gen 1) unless you have low_charge_rate set to on.

Bit surprising there was no discharge rate set by Predbat for the discharge period.

Can you get the history of your inverter discharge rate from home assistant and post that?

springfall2008, 2024-03-23:
It seems like your system isn't keep up, it's taking more than 5 minutes per run, I'd be inclined to reduce your plan update rate to say every 15 minutes or try fast mode enabled.

```
2024-03-22 19:11:25.675079 WARNING AppDaemon: Queue size for thread thread-1 is 346, callback is 'run_time_loop() in pred_bat' called at 2024-03-22 19:10:46+00:00 - possible thread starvation
2024-03-22 19:11:25.675729 INFO AppDaemon: --------------------------------------------------
2024-03-22 19:11:25.676291 INFO AppDaemon: Threads
2024-03-22 19:11:25.683239 INFO AppDaemon: --------------------------------------------------
2024-03-22 19:11:25.690124 INFO AppDaemon: Currently busy threads: 1
2024-03-22 19:11:25.698062 INFO AppDaemon: Most used threads: 1 at 2024-03-10 21:32:53+00:00
2024-03-22 19:11:25.700627 INFO AppDaemon: Last activity: 2024-03-10T21:32:53+00:00
2024-03-22 19:11:25.702881 INFO AppDaemon: Total Q Entries: 346
2024-03-22 19:11:25.704991 INFO AppDaemon: --------------------------------------------------
2024-03-22 19:11:25.713734 INFO AppDaemon: thread-0 - qsize: 0 | current callback: idle | since never, | alive: True, | pinned apps: ['hello_world']
2024-03-22 19:11:25.721531 INFO AppDaemon: thread-1 - qsize: 346 | current callback: run_time_loop() in pred_bat | since 2024-03-22T19:10:46+00:00, | alive: True, | pinned apps: ['pred_bat']
```

bigsmelly, 2024-03-23:
@springfall2008 Thanksf for looking - so what i'm hearing here is that the box being woefully underpowered is causing these issues- i.e.  it's not getting around the loop fast enough, so not being able to send all the required messages to givTCP before it starts again.

 For info, it's a Intel NUC NUC5CPYH, 8GB and a Celeron 3050 2 core. 

It's also very slow when interacting/ changing options, so off to ebay to find a replacement.

@gcoan Also thanks for looking -   For info, logile attached 

[log_inv_1.log.2024-03-21.txt](https://github.com/springfall2008/batpred/files/14733557/log_inv_1.log.2024-03-21.txt)

gcoan, 2024-03-23:
I have an Intel i5-7400 2.4ghz 16gb lenovo m710q which is a decent spec tiny PC for running home assistant 

If you don't want to use it as a PC as well, a pi 5 is fast and very low power

bigsmelly, 2024-03-25:
@gcoan  how noisy is the lenovo? 

I'll close this defect now as the mechanism is understood

closed 2024-03-25 by bigsmelly

gcoan, 2024-03-25:
> @gcoan  how noisy is the lenovo? 
> 
> 
> 
> I'll close this defect now as the mechanism is understood

@bigsmelly there is no noticeable sound from it most of the time.  Occasionally you hear the fan if its working hard but that's infrequently.

It does consume about 35w so that adds up over a day.  That's my only minor criticism. Bought off Ebay and otherwise quite happy with it.

RobinCu, 2024-03-25:
@bigsmelly if noise is a concern, I can recommend a Dell Wyse Thin Client (as it has passive cooling) like this: https://www.ebay.co.uk/itm/256141475616
Mine sits behind my monitors. When I checked consumption, it drew about 5W, and Predbat calc runs take an average of 25 seconds since the multithread enhancement Trefor made.

gcoan, 2024-03-25:
Its a while ago now I bought my PC, but I did look up the minimum recommended to run HAOS on, I think it was 4Gb and an i5-5400 or something like that.

I deliberately bought an over-spec device as I wasn't to have some future growth in it.

There's so many variants of computer processor around it was hard to know what to choose, I did find if I did a google search of "i5-5700 vs celeron 3050" you can find websits that compare the processor speeds.
e.g. my HA machine is 18 months newer procesor than yours and 234% faster

Thin client machines are great for noise and power.  I just couldn't find one at a decent spec when I was looking for one so went with a tiny PC instead