springfall2008/batpred #891: Predbat - Telsa Powerwall 2 integration

springfall2008, 2024-03-28:
### Discussed in https://github.com/springfall2008/batpred/discussions/887

<div type='discussions-op-text'>

<sup>Originally posted by **mtaylor819** March 28, 2024</sup>
Hi there,

I have a Solar array going into a Powerwall 2 via a Solax inverter. I am trying to get Predbat to charge the powerwall using the solar predication with octopus agile. I am struggling lol !!

I have tried the following config but I can't get predbat to start:

```
# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
    - "£"
    - "p"

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # XXX: This is a configuration template, delete this line once you edit your configuration
  #template: True

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today:
    - sensor.my_home_load_power
  import_today:
    - sensor.my_home_site_import
  export_today:
    - sensor.my_home_site_export
  pv_today:
    - sensor.my_home_solar_import

  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast:
  #  - predheat.heat_energy$external
  #
 
  #
  # Controls/status - must by 1 per inverter
  #
  num_inverters: 1
[... 272 lines cut ...]
2024-03-28 16:15:01.554957 INFO pred_bat: Historical data totals for days [7] are [0] - min 0
2024-03-28 16:15:01.535993 INFO pred_bat: PV Data for pv_forecast_d4 total 15.45 kWh
2024-03-28 16:15:01.529765 INFO pred_bat: PV Data for pv_forecast_d3 total 17.64 kWh
2024-03-28 16:15:01.523810 INFO pred_bat: PV Data for pv_forecast_tomorrow total 15.59 kWh
2024-03-28 16:15:01.518171 INFO pred_bat: PV Data for pv_forecast_today total 6.55 kWh
2024-03-28 16:15:01.510719 INFO pred_bat: Todays energy import 12.2 kWh export 0.57 kWh cost 200.42 p import 204.2 p export -3.79 p
2024-03-28 16:15:01.462719 INFO pred_bat: Low Import rate found rates in range 6.81 to 30.24
2024-03-28 16:15:01.453209 INFO pred_bat: High export rate found rates in range 4.39 to 16.44
2024-03-28 16:15:01.443515 INFO pred_bat: Rate thresholds (for charge/discharge) are import 31.950000000000003p (0.0) export 4.33p (0.0)
2024-03-28 16:15:01.440830 INFO pred_bat: Export rates min 3.83 max 16.44 average 7.64
2024-03-28 16:15:01.431842 INFO pred_bat: Import rates min 6.81 max 32.45 average 15.24
2024-03-28 16:15:01.429328 INFO pred_bat: Rate min forward looking: now 6.81 at end of forecast 6.81
2024-03-28 16:15:01.157713 INFO pred_bat: Standing charge is set to 47.9535 p
2024-03-28 16:15:01.104557 INFO pred_bat: Fetching futurerate data from None
2024-03-28 16:15:01.101755 INFO pred_bat: Current data so far today: load 0 kWh import 12.2 kWh export 0.57 kWh pv 0.01 kWh
2024-03-28 16:15:01.099349 INFO pred_bat: Car charging hold True threshold 6.0
2024-03-28 16:15:00.879876 INFO pred_bat: Found 11520 load_today datapoints going back 7 days
2024-03-28 16:15:00.078917 INFO pred_bat: Predbat mode is set to Monitor
2024-03-28 16:15:00.076701 INFO pred_bat: Configuration: forecast_hours 24 max_windows 48 num_cars 0 debug enable is False calculate_plan_every 10 calculate_fast_plan False
2024-03-28 16:15:00.044350 INFO pred_bat: Autoupdate: Currently up to date

It seems it doesn't like reading the charging times from the array, and I thought it would pick it up from the octopus events ?

Please help !
</div>

springfall2008, 2024-03-28:
Welcome :)

I've not tried powerwall before so its hard to comment on specifics but many other systems do work with Predbat:

charge_start_hour:

"{{ expand('array', state_attr('sensor.octopus_agile_sequence_table', 'import_array')) | map(attribute='timefrom') | list }}"
You can't do this, predbat YAML is not HA yaml. You need to create the helper in HA and then reference it here just as an entity name.

sensor.octopus_agile_sequence_table

What is this?

The charge_start_hour if you set this is meant to be the control of the inverters charging start time (needs to be writable)

mtaylor819, 2024-03-29:
So this is the start time found from a node red function I have that extracts the time from octopus for best charging slot. In the array you get a start and end time.

Can predbat read directly from octopus agile and read the charging times ?

gcoan, 2024-03-29:
> Can predbat read directly from octopus agile and read the charging times ?

Yes predbat uses the octopus integration and directly gets the car charging times from the intégration if you're on octopus intelligent go

mtaylor819, 2024-03-30:
Hi there,

I am octopus agile not intelligent go.

I removed the yaml you mentioned but am now getting the following error:

2024-03-30 22:06:14.607098 INFO pred_bat: ERROR: Exception raised
2024-03-30 22:06:14.594075 INFO pred_bat: ERROR: Inverter 0 unable to read charge window time as neither REST, charge_start_time or charge_start_hour are set
2024-03-30 22:06:14.592610 INFO pred_bat: Inverter 0 SOC: 0.0 kW 0 % Current charge rate 2600.0 w Current discharge rate 2600.0 w Current power 0.0 w Current voltage 52.0
2024-03-30 22:06:14.591166 INFO pred_bat: Inverter 0 with soc_max 10.0 kWh nominal_capacity 10.0 kWh battery rate raw 2600.0 w charge rate 2.6 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 3.6 kW export limit 3.6 kW reserve 4.0 % current_reserve 4.0 %
2024-03-30 22:06:14.589131 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-03-30 22:06:14.587154 INFO pred_bat: Inverter 0: Type GE GivEnergy)
2024-03-30 22:06:14.556466 INFO pred_bat: Today's load divergence 100.0 % in-day adjustment 100.0 % damping 0.95x, Predicted so far 0 kWh with 0.0 kWh car excluded and 0.0 kWh import ignored and 0 forecast extra, Actual so far 0 kWh with 0.0 kWh car excluded and 0.0 kWh import ignored
2024-03-30 22:06:14.540668 INFO pred_bat: Historical days now [7] weight [1]
2024-03-30 22:06:14.404139 INFO pred_bat: WARN: Historical day 7 has no data, unable to fill gaps normally using nominal 24kWh - you should fix your system!
2024-03-30 22:06:14.400840 INFO pred_bat: Historical data totals for days [7] are [0] - min 0
2024-03-30 22:06:14.377353 INFO pred_bat: PV Data for pv_forecast_d4 total 10.23 kWh
2024-03-30 22:06:14.370412 INFO pred_bat: PV Data for pv_forecast_d3 total 13.42 kWh
2024-03-30 22:06:14.362951 INFO pred_bat: PV Data for pv_forecast_tomorrow total 12.68 kWh
2024-03-30 22:06:14.356588 INFO pred_bat: PV Data for pv_forecast_today total 20.57 kWh
2024-03-30 22:06:14.350770 INFO pred_bat: Todays energy import 10.1 kWh export 2.56 kWh cost 167.77 p import 184.9 p export -17.14 p
2024-03-30 22:06:14.296842 INFO pred_bat: Low Import rate found rates in range 8.9 to 31.12
2024-03-30 22:06:14.286845 INFO pred_bat: High export rate found rates in range 5.37 to 16.14
2024-03-30 22:06:14.276165 INFO pred_bat: Rate thresholds (for charge/discharge) are import 31.24p (0.0) export 5.23p (0.0)
2024-03-30 22:06:14.273974 INFO pred_bat: Export rates min 4.73 max 16.14 average 7.79
2024-03-30 22:06:14.264916 INFO pred_bat: Import rates min 8.9 max 31.74 average 15.6
2024-03-30 22:06:14.262803 INFO pred_bat: Rate min forward looking: now 8.9 at end of forecast 8.9
2024-03-30 22:06:13.963073 INFO pred_bat: Standing charge is set to 47.9535 p
2024-03-30 22:06:13.906930 INFO pred_bat: Fetching futurerate data from None
2024-03-30 22:06:13.904765 INFO pred_bat: Current data so far today: load 0 kWh import 10.1 kWh export 2.56 kWh pv 0.0 kWh
2024-03-30 22:06:13.902596 INFO pred_bat: Car charging hold True threshold 6.0
2024-03-30 22:06:13.700696 INFO pred_bat: Found 11520 load_today datapoints going back 7 days
2024-03-30 22:06:12.972217 INFO pred_bat: Predbat mode is set to Monitor
2024-03-30 22:06:12.970378 INFO pred_bat: Configuration: forecast_hours 24 max_windows 48 num_cars 0 debug enable is False calculate_plan_every 10 calculate_fast_plan False
2024-03-30 22:06:12.953371 INFO pred_bat: Autoupdate: Currently up to date
2024-03-30 22:06:12.952083 INFO pred_bat: Predbat /config/apps/predbat.py version v7.16.10 currently running, latest version is v7.16.10 latest beta v7.16.10
2024-03-30 22:06:12.417857 INFO pred_bat: --------------- PredBat - update at 2024-03-30 22:06:12.413800+00:00 with clock skew 0 minutes, minutes now 1325
2024-03-30 22:05:57.339803 INFO AppDaemon: App initialization complete
2024-03-30 22:05:57.337361 INFO pred_bat: Balance inverters will run every 60 seconds (if enabled)
2024-03-30 22:05:57.332435 INFO pred_bat: Predbat: Next run time will be 2024-03-30 22:10:00 and then every 300 seconds
2024-03-30 22:05:57.330376 INFO pred_bat: Watch list []
2024-03-30 22:05:57.229806 INFO pred_bat: Note: Disabled configuration item plan_debug
2024-03-30 22:05:57.228133 INFO pred_bat: Note: Disabled configuration item forecast_plan_hours

Config below:

`# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
    - "£"
    - "p"

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # XXX: This is a configuration template, delete this line once you edit your configuration
[... 266 lines cut ...]
  write_and_poll_sleep: 2
  has_time_window: False
  support_charge_freeze: False
  support_discharge_freeze": False

  # Services to control charging/discharging
charge_start_service:
  service: number.set_value
  data:
    value: "100"
  target:
    entity_id: number.2_belcombe_close_backup_reserve
  option: "Charge from Solar Power and Grid"
charge_stop_service:
  service: number.set_value
  data:
    value: "16"
  target:
    entity_id: number.2_belcombe_close_backup_reserve
  option: "Charge from Solar Power"
#discharge_start_service:
#   service: select.select_option
#    entity_id: "select.solaredge_i1_storage_command_mode"
#    option: "Maximize Self Consumption"
`

mtaylor819, 2024-04-03:
Just wondered if anyone had any ideas ?