springfall2008/batpred #893: Different times for inverter and appdaemon

SwiftRR, 2024-03-29:
*I have experienced status reports and warnings in my predbat.log related to unsynchronised times between my inverter time and appdaemon time. 
Could these be connected to my automatic upgrade to Predbat v7.16.10 ?

**Expected behavior**
Times synchronised and no warnings

**Predbat version**
v7.16.10

**Environment details**

- Inverter and battery setup
GE AC coupled 3.0 connected to two GE 8.2 kWh batteries
- Standard HAOS installer or Docker
Standard HA on raspberry pi 4 run from a 512 GB SSD

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.
Screengrab of status errors:
 
[image]

**Log file**
Predbat.log attached:
[predbat.log.txt](https://github.com/springfall2008/batpred/files/14808066/predbat.log.txt)

My appdaemon.yaml was installed automatically on 25th January 2024 when I installed the combined appdaemon/predbat set up. I have attached a copy of this file.
[appdaemon.yaml.txt](https://github.com/springfall2008/batpred/files/14808099/appdaemon.yaml.txt)

Note that the key directories are different from those on the documentation. I have 
  app_dir: /config/apps
    filename: /config/predbat.log
I only get logs numbered up to 3 whereas my file shows 9.

The path to appdaemon.yaml is:
 /addon_configs/46f69597_appdaemon-predbat/appdaemon.yaml

The path to apps.yaml is:
/addon_configs/46f69597_appdaemon-predbat/apps/apps.yaml
For completeness, I have also attached my apps.yaml file
[apps.yaml.txt](https://github.com/springfall2008/batpred/files/14808119/apps.yaml.txt)

Is this something that could be 'fixed' using Clock skew. This extract is from my apps.yaml:
 # Clock skew adjusts the Appdaemon time
  # This is the time that Predbat takes actions like starting discharge/charging
  # Only use this for workarounds if your inverter time is correct but Predbat is somehow wrong (AppDaemon issue)
  # 1 means add 1 minute to AppDaemon time, -1 takes it away
  clock_skew: 0

Predbat seems to be working fine but the warnings are a worry.

Thanks

Rob

gcoan, 2024-03-29:
If you check on the Givenergy portal your inverter clock should be similar to the clock on your home assistant device. 

The fact that you are not getting these warnings that frequent makes me think that the clocks are on similar times but what's happening is that GivTCP is not getting regular updates from the inverter. 

Have a look at the GivTCP log, I think you'll find comms errors reported around the time predbat reports the clocks being misaligned.

Could be an intermittent home network issue, or sometimes just restarting GivTCP is enough. 

You shouldn't need to use clock skew unless the clocks are permanently misaligned and you're getting this error frequently in the logs. 

On the directories the logfile and the apps.yaml are in, they are in different places depending on whether you have used the app daemon or predbat appdaemon combined install but the documentation should make this clear. If you can tell me where it's wrong I'll correct it

SwiftRR, 2024-03-29:
I think you should have all the relevant info in the log file section where I have uploaded predbat.log, appdaemon.yaml, apps.yaml and the HA paths to these files.
When I installed predbat, I used the predbat appdaemon combined install.

I have attached my givtcp log file and my appdaemon-predbat file.
[Givtcp log.txt](https://github.com/springfall2008/batpred/files/14808474/Givtcp.log.txt)
[appdaemon-predbat log.txt](https://github.com/springfall2008/batpred/files/14808475/appdaemon-predbat.log.txt)

I don't know whether there are any clues here – I have no idea what they reveal!
My inverter's connection to my wi-fi is labelled as 'Excellent'. On the GE portal, the time is set automatically on a regular basis although it isn't actually shown on the remote control area.
But in any data screens (battery, pv, etc), the time is correct. 

Rob

gcoan, 2024-03-29:
I'm only on my phone at the moment so it's hard to read the predbat logfile as its so wide. 

But looking at the GivTCP log I count 8 errors in the 7 hour period it covers. The errors are all timeout / failed to decode the inverter output.
There isn't a GivTCP error at the same time as the predbat error you got. 

What frequency do you have the GivTCP polling set to in the GivTCP add on config? If it's set too high the inverter can get overloaded or if too low you'll get timing issues. 
15-30 seconds are a good range of values.

SwiftRR, 2024-03-29:
Is this the polling setting?: (It is enabled)
[image]

SwiftRR, 2024-03-29:
And for the set, here is my mqtt log. It looks strange!
[mqtt log.txt](https://github.com/springfall2008/batpred/files/14808882/mqtt.log.txt)

I have had enough with this for today ..........

Rob

gcoan, 2024-03-29:
Hi Rob

The polling frequency in GivTCP is fine. 30 seconds means GivTCP will retrieve your inverter data every 30 seconds.

The mqtt log looks fine, just normal chatter in that.

Try giving mosquitto a and GivTCP a restart and see if that cures the errors in the GivTCP log

SwiftRR, 2024-03-29:
I have updated mqtt to latest version (I worry about doing this in HA in case it breaks things).
I have restarted both mqtt and givtcp.
The givtcp log looks clean. I can't see an error (yet) and 

I have attached the latest!
[givtcp log at 22-43.txt](https://github.com/springfall2008/batpred/files/14809018/givtcp.log.at.22-43.txt)

Rob

SwiftRR, 2024-03-29:
givtcp has been running for a while now and I have now seen 2 errors, the same as before (whatever this means!):

> AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response'), <traceback object at 0x7f9d4a8b40>)

Rob

closed 2024-09-01 by springfall2008