springfall2008/batpred #898: Clocks changing

slopemad, 2024-03-30:
Is there a potential issue with Predbat with clocks going forwards tonight?  I ask because:

[image]

There is, of course, no actual 1am and 1:30 am slot, and the 2am localtime price of 18.04 is showing at 1am...  Similarly, the 4pm peak price is showing at 3pm...

gcoan, 2024-03-30:
I saw that the peak prices were starting at 3pm as well. I guess it makes sense in terms of being X hours away.

I'm assuming that when the clock jumps forward predbat will just jump forward and adjust to the new time. Same as if the PC was switched off for an hour 

When the clocks went back in October the octopus integration had duplicates of 1am and it all seemed to work out OK

gcoan, 2024-03-30:
Mmm, ok so there is some strange behaviour in the predbat plan

The Octopus agile rates don't actually have a 1:00 and 1:30 slot

[image]

But predbat doesn't expect that and sets the 1:00 price to the 2:00 price from octopus 

[image]

Which kind of makes sense even if the times in the plan are wrong

But only a once a year issue

springfall2008, 2024-04-21:
I think it's fine as the calculations are done internally based on time since midnight, its just the output plan doesn't localise the times?

springfall2008, 2024-11-16:
Closing this old ticket, please open a new one if you have an up to date question

closed 2024-11-16 by springfall2008