springfall2008/batpred #899: SunSynk inverters

xcentric7881, 2024-03-31:
Does batpred work with Sunsynk inverters? They're not on your list of supported ones - what do I need to know from sunsynk specs to tell/to get this to work with them?  They are also sold as Deye inverters, and quite popular so a decent base of potential users.....

https://www.sunsynk.org/3-6kw-5kw-ecco

thanks.

iendicott, 2024-04-07:
Yes I would be interested in that too.

springfall2008, 2024-04-12:
Do you have a home assistant integration up and running and are the controls documented?

iendicott, 2024-04-12:
I do have it up and running. As to documentd no, but there are comments on the oroginal YAML for the ESP module if that helps?

For example

` - platform: modbus_controller                   #243 Select Energy Patern
    use_write_multiple: true
    modbus_controller_id: sunsynk
    name: "${friendly_name} Energy Pattern"
    id: sunsynk_select_energy_pattern
    address: 243
    value_type: U_WORD
    optionsmap:
      "Battery first": 0
      "Load first": 1   
         
  - platform: modbus_controller                   #244 Select Work Mode
    use_write_multiple: true
    modbus_controller_id: sunsynk
    name: "${friendly_name} Work Mode"
    id: sunsynk_select_work_mode
    address: 244
    value_type: U_WORD
    optionsmap:
      "Selling First": 0
      "Zero Export + Limit to Load Only": 1
      "Limited to Home": 2    

  - platform: modbus_controller                  #280 Select Grid Peak Shaving
    use_write_multiple: true
    modbus_controller_id: sunsynk
    name: "${friendly_name} Grid Peak Shaving"
    id: sunsynk_esphome_select_grid_peak_shaving`

bensebborn, 2024-04-14:
I have Sunsynk working via solar assistant. 

Would be great to get this working

Let me know what info I can provide to help.

springfall2008, 2024-04-21:
There's another thread on this here: https://github.com/springfall2008/batpred/issues/966

closed 2024-05-03 by springfall2008