springfall2008/batpred #902: Solcast - multiple arrays

TheSargey, 2024-04-01:
Afternoon folks, Happy Easter.

Just a quick question, if I have a split array setup in Solcast (i.e. Two arrays) does Predbat amalgamate these to assess the forecast or does it just take data from the first array? I've got a feeling mine is just taking data from one array but just wanted to validate my understanding.

Many thanks as always,

Sarge

gcoan, 2024-04-01:
You'd think this'd be easy to answer....

I have two arrays configured in solcast, East facing and West facing.  I was pretty sure that the data predbat gets was for both arrays (or sites as solcast calls them) but confirming this proved hard.

In solcast you enter your account API key, not the site id key.  The two sites are held as sub-entities within your account.

The solcast GitHub documentation doesn't really say, and when I log in to solcast to see what data it predicts and compare that to what I get in HA, I can see graphs of my predicted generation per site (array) but not the actual PV prediction!

However, in HA solcast creates two sensors for the two sites, and looking at today's values they are 37kW and 6kW.  The total solcast prediction is 43kW, so the sum of the two sites.
Looking at the graphs in solcast website I can see the East site is much bigger peak than the West site peak, so whilst I can't see actual generation estimates in solcast, this does seem to match.

Predbat only gets a single (set of hourly) values from Solcast, it doesn't have a breakdown by hour by site/array.

TheSargey, 2024-04-04:
Hi @gcoan - thanks for coming back to me and apologies for the delay.

I do get a bit confused by the solar prediction data in Predbat.  Take the following as an example...

On my dashboard I have gauges setup to show current PV rate (sensor.givtcp_sa_pv_power) and predicted PV using (sensor.solcast_pv_forecast_power_now).  These are generally in tune although can vary a bit.  I've also got a prediction graph for the next few days using the solcast sensors.  In the example below there was a gap in the clouds which meant the generated PV was much higher than the predicted.

[image]

However, when I look at the predplan the predicted PV kWh is usually much lower than both predicted and current generation.  In this example it shows the PV kWh at 0.39 when the predicted was 1000 W and the actual was 2.8kWh

[image]

My assumption is that the predicted PV within the plan is taking in multiple factors (load scaling PV10, Pv scaling etc.) therefore perhaps I need to tweak these values to be more ambitious within the predbat plan?  If so, which parameters do you think are best to adjust to make predbat more optimistic?

I'm guessing that when summer finally comes (please let it come soon!) that this will be pretty much moot and the predictions from solcast will be more consistent, however I'm still unsure as to whether I'm reading the PV kWh in the plan correctly, should this correlate more closely with the kWh being suggested by solcast?

Thanks as always for your time.

Sarge

gcoan, 2024-04-04:
Hi Sarge

There's a couple of things going on here.

Firstly don't get confused between power and energy.

Using the predbat plan example you gave, the predbat prediction for that half hour slot is that the TOTAL PV energy generated will be 0.39kWh.
It says nothing about how the energy in that time period will get to that 0.39kWh, just that over the 30 minutes it will sum up to 0.39kWh.
So you seeing an INSTANTANEOUS power of 2.8kW (note, its 2.8kW not 2.8kWh as its power not power over time) is not incompatible with the predbat prediction.
You could get 2.8kW for 10 minutes and then nothing generated for 20 minutes so the total energy generated would be 2.8*10/60=0.46kWh.

Secondly you are absolutely correct, Predat is blending different solcast forecasts together to come up with the prediction you see in the predbat plan.
Solcast produces a PV50, PV10 and PV90 forecast and the graph you've produced is almost certainly the PV50 forecast which for most people is too high.

Looking at my own historical solcast forecasts and actual generation, I estimate its usually somewhere around 70-80% between the PV10 and PV50 forecasts:
[image]

Coming back to Predbat, it uses a blend of the PV50 and PV10 forecasts https://springfall2008.github.io/batpred/customisation/#scaling-and-weight-options using input_number.predbat_pv_scaling and input_number.predbat_pv_metric10_weight.  I have my pv scaling set to the default of 1 and have increased pv metric 10 to 0.3 to be a bit more pessimistic about solar generation (and more in line with the graph above).

Its best to look at tuning PV based on a number of days of data not just a single day. Today for example PV forecast is 29.6kWh, PV10 is 18.7 and actual generation 10.6kWh, so definitely at the lower end for me.

TheSargey, 2024-04-15:
Hi Geoff,

Thanks for the usual detailed response and my apologies again for the delay, holiday season and all that.

Ah indeed, I shouldn't be comparing kWh with straight kW - I'll keep this in mind going forward.

Interestingly though, looking at a comparable graph to yours, it would seem my Solcast is predicting lower than what is occuring (in most cases) and I'm wondering again if this is because of the way I've split the array.  I have my pv scaling at 1 and pv metric 10 at 0.15 so I'd have thought if it was an issue with predbat calculations then the prediction would be much more optimistic than the actual.

[image]

I'm going to change my Solcast array first I think and see where that gets me, I just feel like it's misinterpreting my solar array since I split it into two and that's affecting the prediction level.  Here's the Solcast prediction over the previous month using the straight Energy tab in HA:
[image]

So without any predbat calculations we can see that Solcast is underestimating the solar generated in most cases across the past couple of weeks so I definitely think I need to take another look at that before trying to change anything in predbat.

Again, many thanks for your time and help as always.

Best,

Sarge

gcoan, 2024-04-15:
Yes the evidence does point towards needing to look at your solcast site configuration - the array orientation and number of panels would be the first thing I'd double check.

On the energy dashboard it shows the PV50 line which is a bit optomistic, and yet you are achieving well over it every day.

Here's my own energy graph for this month, the lightest yellow bars are for my FIT array that I deliberately don't include in my solcast forecast as that array has an AC inverter on and my two GE arrays (darker yellows) have hybrid inverters with batteries - so I don't tell solcast/predbat about the FIT as that solar can't be stored in the batteries only consumed by the house or exported.

If you exclude the lightest yellow, my daily production is slightly below the solcast PV50 prediction - opposite way round to yours
[image]

closed 2024-09-01 by springfall2008