springfall2008/batpred #905: Support for GE inverter using EMC controller

springfall2008, 2024-04-01:
**Is your feature request related to a problem? Please describe.**
Support for multi-inverters controlled by the offical GE EMC module

**Describe the solution you'd like**
Integration into Predbat

**Describe alternatives you've considered**
Disable EMC and use GivTCP

**Additional context**
Discussions ongoing with GE about supporting this controller

There is no local interface so it must be via cloud controls

gcoan, 2024-04-01:
Great to hear you've got a line of communication to GE and an interesting development but personally low priority for me.  

EMC due date - who knows with any realistic certainty?
Likely audience for the EMC will be limited to multi-AC users in the first instance, maybe multi AIO and hybrid in the longer term, but again, when?
Cloud control only, a big turn-off as that then puts ongoing operation/free access etc entirely at the whim of GivEnergy
And then a Q of the affordability of the EMC.

springfall2008, 2024-04-11:
Now rolled out and working

closed 2024-04-11 by springfall2008