springfall2008/batpred #907: Best SOC Keep Not Being Maintained

nbullus, 2024-04-02:
**Describe the bug**
The **Best SOC Keep** setting is meant to keep the battery SOC above the user specified threshold (1.4kWh - 10% SOC in my example).  On numerous occasions I have seen the SOC drop below this threshold and down as low as my 5% **SOC Reserve**.  In the attached example plan, the SOC is forecast to drop below my 10% threshold between 22:00 and 23:30. 

**Expected behavior**
Predbat should not intentionally plan to take the SOC below the **Best SOC Keep** threshold in future slots. The SOC should only drop below this threshold in the current slot if either the demand is higher than forecast and/or solar is lower than forecast.

If the SOC does drop below the threshold due to unexpected changes in load/PV, it should restore the battery level back to the **Best SOC Keep** at the next Charge slot opportunity.

In the attached screenshot the Discharge Slot at 16:00 should have discharged down to 30% rather then 24% to prevent the SOC forecast to drop to 6% later in the plan.

**Predbat version**

7.6.10

**Environment details**

- GivEnergy AIO + Solis string inverter
- Standard HAOS installer

**Screenshots**

[image]

[image]

**Log file**

[Predbat Log.txt](https://github.com/springfall2008/batpred/files/14844726/Predbat.Log.txt)

gcoan, 2024-04-03:
Worth you having a look at the documentation for best_soc_keep and best_soc_min https://springfall2008.github.io/batpred/customisation/#battery-margins-and-metrics-options

best_soc_keep is not a hard battery minimum level, as explained in the documentation its ignored for the forthcoming 4 hours of the plan, I think this might be what's happening.

I am personally seeing predbat doing extra discharges a bit too early (on the fixed agile 15p outgoing tariff) that result in the Soc reducing to empty and unprofitable imports #861.  This might be a different symptom of this.

nbullus, 2024-04-03:
The issue is not with slots in the next 4 hours, but slots which are 20+ hours away.

springfall2008, 2024-11-16:
Closing this old ticket, please open a new one if you have an up to date question

closed 2024-11-16 by springfall2008