springfall2008/batpred #910: appdaemon-predbat add-on log

SwiftRR, 2024-04-03:
**Describe the bug**
My appdaemon-predbat log file contains warnings and an error. 
I have looked back at issue #341 which I think predates the combined appdaemon-predbat add on. #341 does include a screenshot of the appdaemon-predbat log file at the time.

I don't know whether my appdaemon-predbat log file  signals possible inconsistencies in the predbat code.
Or whether these are just 'warnings'.
I have been experienced many strange warnings in log files (See issue #857) and this is the last log I have investigated!

**Expected behavior**
No warnings or error

**Predbat version**
v7.16.10

**Environment details**
Predbat installed on Raspberry pi4 on a large SSD (512gb)

- Inverter and battery setup
Standard HA OS 11.5 (v2014.3.3, freshly upgraded)

**Screenshots**
Attached is a screenshot of the appdaemon-predbat log taken from HACS add-on.
 **Log file**
[image]
Also attached is predbat.log that contains the same times as the warnings in the appdaemon-predbat log. See 2024-04-03 15:06:31 onwards.
[predbat.log 2014-04-03.txt](https://github.com/springfall2008/batpred/files/14857965/predbat.log.2014-04-03.txt)

Rob

gcoan, 2024-04-03:
As long as you have done an update of predbat within the combined predbat-appdaemon install, the predbat code is the same.   The combined install contains an older version of predbat, but once you update predbat, it should operate identically.

Are these errors just after you restarted Home Assistant?

If I look in my own logfile just after restarting HA I see a similar set of errors, from predbat, just erroring on 'export' rather than 'load_today'.

Predbat relies on a number of components - GivTCP, MQTT, HA itself, Solcast, Octopus Energy integration, etc.  When HA restarts these are all busy loading and I can see on my dashboard there's a visible delay whilst the different components get going, establish their connections, etc.

What's I suspect is happening is that predbat starts up, it tries to get data from HA and GivTCP and they're not fully initialised yet so the request fails.

It may be possible to put some more clever 'am I starting up, don't error so enthusiastically' or 'delay 5 minutes after HA startup' in the predbat code,  but as long as Predbat gets going OK and runs fine after these errors I think its safe to ignore them.

My own logfile:

```

s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[18:16:11] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2024-04-03 18:16:18.743796 WARNING PVCalc:: ------------------------------------------------------------
2024-04-03 18:16:18.744673 WARNING PVCalc:: Unexpected error running initialize() for PVCalc:
2024-04-03 18:16:18.744933 WARNING PVCalc:: ------------------------------------------------------------
2024-04-03 18:16:18.746936 WARNING PVCalc:: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
    await utils.run_in_executor(self, init)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/PVCalc.py", line 29, in initialize
    self.collect_sensor_data()
  File "/homeassistant/appdaemon/apps/PVCalc.py", line 50, in collect_sensor_data
    value = int(float(value) * 1000)
                ^^^^^^^^^^^^
ValueError: could not convert string to float: 'unavailable'

2024-04-03 18:16:18.747814 WARNING PVCalc:: ------------------------------------------------------------
2024-04-03 18:36:49.466579 WARNING pred_bat: ------------------------------------------------------------
2024-04-03 18:36:49.466863 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-04-03 18:36:49.468517 WARNING pred_bat: Worker Ags: {'id': '261239f0c1a2424bbf7a82495fdf860c', 'name': 'pred_bat', 'objectid': 'acfa554b062745a0a9982e1986e5752f', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7f1a610ce1d0>>, 'pin_app': True, 'pin_thread': 3, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-3'}}
2024-04-03 18:36:49.470456 WARNING pred_bat: ------------------------------------------------------------
2024-04-03 18:36:49.473997 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 12687, in update_time_loop
    raise e
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 12682, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 11681, in update_pred
    self.fetch_sensor_data()
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 11135, in fetch_sensor_data
    self.rate_export = self.basic_rates(self.get_arg("rates_export_override", [], indirect=False), "export", self.rate_export, self.rate_export_replicated)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 6342, in basic_rates
    rate = self.resolve_arg("rate", rate, 0.0)
                                    ^^^^
UnboundLocalError: cannot access local variable 'rate' where it is not associated with a value

2024-04-03 18:36:49.474307 WARNING pred_bat: ------------------------------------------------------------
```

SwiftRR, 2024-04-03:
Thanks for the rapid response, and for sharing your log which reassures me!
You are absolutely correct that predbat is juggling many components and does an amazing job. But reloading them all will take a little time.
  
After a succession of my usual 'random' status messages this morning, I have had a clear run since 11:30 am.

I restarted HA this pm and took the screenshot of my appdaemon-predbat log immediately afterwards.

Rob

closed 2024-04-11 by springfall2008