springfall2008/batpred #911: Chosen charge slots not optimised

mpartington, 2024-04-04:
**Describe the bug**
Predbat is not selecting the cheapest slots to charge.
I've suspected this behaviour for a while, but yesterday I screenshot the plan and have compared it to what Predbat actually did.

Basically it charged up on the slots times I have highlighted in PURPLE, but some of these (1.30 and 2am particularly) were more expensive than later slots.

2am slot was originally planned as discharge, but Predbat must have changed then plan.

**Expected behavior**
Choose the cheapest slots when multiple are available and hold or discharge (depending on price) for the others

**Predbat version**
V7.16.10

**Environment details**
- Standard HAOS installer 

**Screenshots**
Settings in log file, but
Combine charge slots is off as is any second pass optimisation.

[image]

[image]

**Log file**
[appdaemon (3).log](https://github.com/springfall2008/batpred/files/14861453/appdaemon.3.log)

mpartington, 2024-04-04:
Obviously it's got this almost right and the cost impact is small, but it appears to point to something not quite right in the optimisation.

Mikebert4, 2024-04-21:
I'm seeing similar behaviour.

Here's my plan for today - PredBat is charging in the red-bordered slots, but it would be more cost effective to hold over these and charge in the green-bordered slots.

[image]

Settings wise, I'm identical to OP, excepting that I have "set_dicsharge_during_charge" and "set_discharge_freeze" both as False.

mpartington, 2024-04-21:
Mine did similar last night. Charged at 23.30, rather than holding to 02.30.

It did occasionally generate a plan that did hold, but seemed to revert and favour to the earlier charge option

[image]

mpartington, 2024-04-25:
Much better and more recent example. The plan was never choosing the lowest rates.tomcharge at, but then it must have got even worse after I went to bed, as it charged up at almost the highest rates in the period (23.30 to 00.30), rather than using those highlighted in green.

It needs to hold until later, as per my markup.

[image]
[image]
[image]

mpartington, 2024-04-25:
Here's the ned debug files for the period. Unfortunately the main log has overwritten (as running debug it quickly fills and overwrites)
[predbat_debug_01_00_00.zip](https://github.com/springfall2008/batpred/files/15105332/predbat_debug_01_00_00.zip)

gcoan, 2024-04-25:
> Unfortunately the main log has overwritten (as running debug it quickly fills and overwrites)

If you edit the AppDaemon.yaml config file (in /addon_configs/... appdaemon - depending on which AppDaemon approach you have used)

Can change the number and size of log files retained:

```yaml
# write log records to a file, retaining 20 versions, rather than the standard appdaemon log
logs:
  main_log:
    filename: /homeassistant/appdaemon/appdaemon.log
    log_generations: 20
    log_size: 5000000
```

mpartington, 2024-04-25:
[predbat_debug_info.zip](https://github.com/springfall2008/batpred/files/15108412/predbat_debug_info.zip)

mpartington, 2024-05-04:
This has been fixed

closed 2024-05-04 by mpartington