springfall2008/batpred #918: input_number.predbat_car_charging_plan_max_price wont go negative

SteveJack34, 2024-04-05:
With tomorrows extend period of negative on Agile thought I would set input_number.predbat_car_charging_plan_max_price to -2p to get the most benefit and charge the car at the best prices. 
I have to type a number into input_number.predbat_car_charging_plan_max_price I can't use the up and down arrows and when it chooses when to charge it still seems to be using 0p as the minimum price.

[image]

gcoan, 2024-04-05:
As a quick fix, change line 472 of predbat.py, the "min" value to say -10:

```yaml
   {
        "name": "car_charging_plan_max_price",
        "friendly_name": "Car Charging Plan max price",
        "type": "input_number",
        "min": 0,
        "max": 99,
        "step": 1,
        "unit": "p",
        "icon": "mdi:ev-station",
        "default": 0,
    },
```

SteveJack34, 2024-04-05:
Hi Trefor
I've tried typing in -2 but its still selecting below 0
[image]

SteveJack34, 2024-04-05:
Now got the value going below 0. Something still not right. I've set max_price to 1p so it should use the 03:00 slot but hasn't selected it.
[image]

gcoan, 2024-04-06:
i was wondering if it wasnt selected because in total the other slots would be enough to fill the car ? The 3:00 is comparatively expensive

springfall2008, 2024-04-11:
Fix on main for testing

springfall2008, 2024-04-13:
Fix now released

closed 2024-04-13 by springfall2008