springfall2008/batpred #927: Support for Growatt inverters?

SwiftRR, 2024-04-08:
Are you looking to expand predbat support into Growatt?
Is there an api?

This enquiry is on behalf of a friend who is interested in predbat but has Growatt inverter and battery.

Thanks

Rob

springfall2008, 2024-04-12:
I don't own a Growatt so I don't know.

Is Growatt integrated into home assistant and if so can you provide documentation of the controls?

MuadDib007, 2024-08-06:
Growatt control via the [Grott addon from muppet3000](https://github.com/muppet3000/homeassistant-grott) is limited as far as I know and works best when combined with an on-premise [Grott by Johan Meijer](https://github.com/johanmeijer/grott) which listens in on the packages between your inverter and Growatt.
This setup is far more stable than the cloud version of the "default" Growatt integration in Home Assistant.

However growatt inverters can be Modbus controlled as well.
There are differences in the registers used by different models.

Personally I have a MOD7000TL3-XHBP which I can control via Modbus and am willing to make available for development.
At the moment I still have a fixed price contract but when that ends I would like to start using PredBat.

sparkyhughes, 2024-12-18:
Solar-Assistant can integrate with Growatt inverters. Small expense but worth it. Can hook into Home Assistant via MQTT.

PhillyGilly, 2025-02-12:
I am chugging away on this. https://github.com/PhillyGilly/Predbat-for-Growatt-SPH/
You can talk to the inverter via its RS485 port using an interface device - exactly the same idea as the supported sofar2mqtt inverter - but using ESPHome platform.
Lile other people, it's not my inverter (I went GivEnergy exactly because of its connectivity) so I'm waiting to be able to test my interface. But once that's working, I should be ready to move forward.
One thing that I'm not clear about is the right way to deal with differences in data structure between systems. For example charging slot times. As far as I can see from mqtt explorer GivTCP is working with **Charge_start_time_slot_1 = 11:30:00** whereas my growatt integration is creating **Battery BatteryFirst Timeslot 1 start Hour  = 11** and **Battery BatteryFirst Timeslot 1 start Minute = 30**. One approach is to write ESPHome automations to update these two registers on reciept of an mqtt message, which looks do-able, but is that the best approach?

gcoan, 2025-02-12:
> As far as I can see from mqtt explorer GivTCP is working with Charge_start_time_slot_1 = 11:30:00 whereas my growatt integration is creating Battery BatteryFirst Timeslot 1 start Hour = 11 and [...]

There's a few ways this could be done

Obviously could change Predbat to deal with different data structure format and write the data the Groatt inverter needs natively. It is possible but adds to the Predbat code complexity especially if lots of different formats get added for other inverter types over time

Could use ESPHome to do the conversion

Could create Home Assistant automations to convert when a dummy helper input_datetime (that predbat writes to)

Could use template sensors that will be automatically triggered when the dummy input_datetime changes. 

The difficulty is how you get the data into ESPhome.  If ESPhome can read a HA sensor/pair of sensors the last might be the easiest otherwise you'll have to do something inside the 'ESP world'

johanzander, 2025-02-12:
If it's of any use for anyone, I've extended the official Growatt Home Assistant integration with support for Growatt TLX inverters and APX batteries. Many have taken the Solar Assistant approach, but I pursued the growatt API path, and it's been running stable in my system for many months now. My intention was to contribute it back to the community, but the maintainers wanted me to submit small Pull Request iteratively, but I've made a bit of refactorisations and I'm not sure if/when I'll find the time to contribute it back. Anyway here it is: https://github.com/home-assistant/core/pull/133319.

Once I was done with the HA Growatt integration, my idea was to use batpred for control, but I ended up developing my own simple control algorithm while exploring and understanding how to actually control my Growatt system. The biggest challenge is/was that the Growatt settings (especially TOU) are really not designed for automatic hourly control, and there are so many limitations I'm not sure how to make this generic for batpred use.