springfall2008/batpred #929: How to handle a firmware update with Predbat?

jeldwendean, 2024-04-09:
There's a new BMS version for the GivEnergy AIO. If I want to update, do I need to do anything on the Predbat side?

gcoan, 2024-04-09:
You shouldn’t need to, but might be worth holding off or asking on the givenergy forum to check whether anyone else has had any experience of it with givtcp.Predbat itself wouldn’t normally notice any changes, its all hidden behind givtcp.A BMS version can often require a full battery calibration when it is installed so you might want to think about the timing of this, when the discharge then charge runs.Predbat will notice that a battery calibration is underway, will pause its scheduled battery activity and set battery discharge and charge to maximum values.When I had new BMS firmware installed (not on an AIO) this was about lunchtime so it was going to discharge into the peak then recharge in the peak - not ideal.I set predbat to read only, then manually adjusted the battery discharge rate so the battery would last me through the peak.  I then set the charge rate to zero, ran on grid through the evening, then set the charge rate back up to max to do the full battery charge overnight.Cheers

gcoan, 2024-04-10:
> Ok great, if Predbat knows when a calibration is taking place then I'll kick the upgrade off tonight 

Yes Predbat knows when the calibration is taking place

https://springfall2008.github.io/batpred/faq/#warn-inverter-is-in-calibration-mode

Good luck

closed 2024-04-20 by springfall2008