springfall2008/batpred #933: Appdaemon-predbat install

jaseHA, 2024-04-10:
**Describe the bug**
I am not seeing the full predbat config appear in apps.yaml file, see screen shot

**Expected behavior**
No errors when loading and starting add-on (not this time) just noting populated in apps/apps.yaml file other than see screen shot 2
[image]
[image]

**Predbat version**
7.16.7
xxxx

**Environment details**

- Giv AIO
- Standard HAOS Pi4

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

gcoan, 2024-04-10:
That's very strange

You can just copy the standard template apps.yaml file from https://github.com/springfall2008/batpred/blob/main/apps/predbat/config/apps.yaml to replace the very empty one that has been installed

jaseHA, 2024-04-10:
I have done this but the install failed to produce any Predbat entities. here is the log file!
[image]

gcoan, 2024-04-10:
There's something weird with your install. The logfile shows that the apps.yaml is not finding the predbat.py file

I just installed the appdaemon-predbat add-on myself (I don't normally use it), started the add-on and it downloaded apps.yaml and predbat.py correctly from github

My directory structure looks like this (this is in visual studio server), does it match yours?

[image]

And this is what my logfile looks like for comparison, shows it loading predbat OK
[predbat (1).log](https://github.com/springfall2008/batpred/files/14936412/predbat.1.log)

jaseHA, 2024-04-10:
Thanks for the reply. I have no PredBat.py file, only a hello.py file.

gcoan, 2024-04-10:
You could manually copy the predbat.py file off GitHub but I'm wondering what else is wrong with this install. 

Suggest deleting the predbat app daemon add on, installing it again and then starting it up. The apps.yaml and predbat.py are downloaded in the initial startup

jaseHA, 2024-04-10:
I think that makes sense, I’ll delete the add on and start again clean. Thanks for looking into this for me.

jaseHA, 2024-04-11:
Sorry, I still can’t get this to load. Making some progress but now it is not making the PredBat entities. It’s making the PredBat.py. I’ve not deleted and reinstalled 3 times deleting all the files that the install makes each time. I have restarted pi4, must be missing something simple but following latest YouTube install video closely!

gcoan, 2024-04-11:
you've got the predbat.py file, good,  and sounds like its starting up ok without errors that you had last time

have you configured apps.yaml to your system requirements?

once you do this and comment out or remove the template=true line predbat will create the entities. 

https://springfall2008.github.io/batpred/install/#configuring-predbat

jaseHA, 2024-04-11:
here are the error logs from the Add-on and the predbat log file. if it helps you spot anything obvious!
[image]
[image]

gcoan, 2024-04-11:
Your energy rates are not setup properly. 

You are with octopus I see.

Here's the relevant part of the documentation https://springfall2008.github.io/batpred/energy-rates/

Have you enabled the octopus integration events?

jaseHA, 2024-04-12:
I thought it would pick the rates up automatically, but I can add them.

I have 2 automations running, DFS and another that picks up extra intelligent Octopus slots for charging, using the integrations for Bottle cap Dave and Mega Kid. I guess I can disable them now.

gcoan, 2024-04-12:
predbat will pick up the rates from the Bottlecap dave integration automatically, but the error in the predbat log file is that there is no current or previous rates in the entities its found.

So either the integration isn't setup right, the entities are not enabled (see instructions), or predbat is choosing the wrong entity names.

Can you show what's in those entities that predbat says are empty

jaseHA, 2024-04-12:
I have enabled the events in BCD’s integration that were disabled and now have all PredBat entities available. Thank you for the advice and getting me started. Coffee on its way to you.

gcoan, 2024-04-21:
closing as now resolved

closed 2024-04-21 by gcoan